bluerange-io/bluerange-mesh #4: Implement really reliable messages (end-to-end acknowledged)

mariusheil, 2015-08-18:
Some messages must be implemented in a way that is bullet-proof. We want the receiving device to acknowledge the message reception. We should therefore send a request handle for that message to keep track of it.

junaediadi, 2016-05-06:
Dear Marius,
Is the reliable and unreliable message has been implemented in the new version?
I tried to test it by change the value to be "true" or "false"
cm->SendMessageToReceiver(NULL, _packet, dataLength, true?false);

I think that sending message using reliable connection will take time longer than unreliable because it needs to make certain guarantees about the successful transmission.Is it right?

However,In my experiment,the result of time to send data between reliable and unreliable is little bit similar.

And also,how many packet that send by using fruity mesh in every advertising connections?
Based on [this book](https://www.safaribooksonline.com/library/view/getting-started-with/9781491900550/ch01.html),we can have 6 packets with payload 20 bytes of each packet. 

Based on your [pages](https://github.com/mwaylabs/fruitymesh/wiki/The-Algorithm-in-Detail), the payload of each packet is 16 bytes.

lets assume that we used 500 ms for connectable advertising interval. As a result,we can get minimum 192 bytes for 1 second connection. Is it right?

Sincerely yours,

Junaedi Adi P

mariusheil, 2016-05-09:
Hi,

I've just tested with my send command: "send u 20" and "send r 20" (unreliable and reliable). And the unreliable one takes longer. The reliable one uses reliable packets of the BLE protocol and waits for an ack of the receiver. This takes a lot longer because the ack will only arrive in the next connection event. Sending multiple packets per connection event has been improved with the latest S130 v2 but you should read a little in the nordic devzone for how many packets are supported. It depends a little on the configuration and on who is Peripheral and who is Central. But you can easily run some tests. Throughput is really dependant on the link quality and will degrade a lot if the connection is bad.

junaediadi, 2016-05-11:
Thanks for your reply.
I still don't get the idea why unreliable takes longer.
If I'm not mistaken, unreliable is communication that doesn't need ACK. So they will send the data without waiting the ACK from the receiver. And why unreliable takes longer than reliable?

I tried with NRF51DK with the latest fruity mesh that only transmit 1 packet per connection event.
I used connection interval 100 ms and used reliable communication with the command "datal":
20 bytes: need times around 110 ms
30 bytes: need times around 503 ms

mariusheil, 2016-05-12:
Hello,

I can't really reproduce your results. I am using the github/master and
used the command "update_iv 0 1000" to switch to a connection interval of 1
second. The command "send u 20" takes about 9 seconds to transmit its
unreliable packets (without ACK) and sends multiple packets per connection
interval which can be observed in the terminal. The "send r 20" command
that sends WRITE_REQ (reliable) instead of WRITE_CMD takes way longer and
can only send one packet every one or two connection intervals. It finishes
after about 40 seconds.
When you send a packet that is larger than 20 bytes payload, it will always
be sent reliable. This is enforced by the implementation because there is
no information transmitted about the order in which the packets arrive,
which means they have to arrive in the exact same order without any loss.

You can try to use "debug conn_data" to get some log output for sending and
receiving packets.

Hope that helps :-)

junaediadi, 2016-05-12:
Dear Marius,

It's really clear now.

I send the data using datal (which means the data larger than 20,so why the it took times longer because they will send it using reliable,unless I did setting to be unreliable).

I also tried using send r and send u,the results is similar to you. I never reach maximum 6 packets per connection intervals. I did  only 3 packets per intervals is the maximum. However, it is better than previous Softdevice S120.

The best idea to send bigger data is splitting the data to be 20 bytes before sending to receiver,as you said before.

Sincerely yours,
Junaedi Adi P

closed 2017-08-09 by mariusheil