bluerange-io/bluerange-mesh #5: Bootloader for precompiled build

Anwain, 2015-08-19:
Hey,

Thanks for starting this! Which bootloader did you use for the precompiled build?
Also, would this work with nRF51 beacons?

mariusheil, 2015-08-19:
Hi,

the precompiled 0.1.7 build does not use a bootloader yet. It uses some startup code from this library: https://github.com/I-SYST/EHAL that was provided with the tutorial that I've used to set up my development environment: http://embeddedsoftdev.blogspot.de/p/ehal-nrf51.html
Future releases should work with a bootloader that enables firmware updates and the DFU module should be extended to communicate with that bootloader.
What exactly do you mean with nRF51 beacons? As long as it is a nRF51 chipset with at least IC revision 3, it can run the S130 SoftDevice and should run fruitymesh (some optimizations are needed for the 16kb ram version).

closed 2015-08-19 by mariusheil

Anwain, 2015-08-30:
Okay, here is the beacon kit https://www.nordicsemi.com/eng/Products/Bluetooth-Smart-Bluetooth-low-energy/nRF51822-Bluetooth-Smart-Beacon-Kit
It requires a bootloader during the firmware update. I'll have another go and see how it works

mariusheil, 2015-08-31:
I see, you are not flashing it via the JTAG connector but over the air. You'll have to add the bootloader then, this is something I have not done yet.