bluerange-io/bluerange-mesh #10: gateway node for internet routing

robomotic, 2015-08-21:
Hello,
excellent work and I was looking at your future milestones.
Do you plan also to implement a node that will act as a bridge i.e. use a mobile phone as a gateway to stream the mesh data online?

Cheers!

mariusheil, 2015-08-22:
Hi,

a gateway is also planned in both directions. The plan is to design a protocol that handles GATT procedures over the mesh. This would allow the mesh to contact GATT devices without mesh functionality and provide access to their attributes ober the network. A mesh-gateway node would then set up a "fake" GATT server and route changes to the selected device. This involves a few problems for acknowledging changes and I guess notifications would pose a serious problem as well, but it's planned to include some of these features.

Streaming the mesh data through a mobile phone should be rather easy. One could implement a module that provides a simple GATT service and then configures a node in a way that it has a free connection left. This would probably be a connection where the node acts as a central and the mobile phone as a peripheral because we have three of these connections. An app could then just stream the received data through its mobile connection. I am not sure if that is on our roadmap, but I'll try to get feedback on this.

robomotic, 2015-08-23:
Hello Marius,
that is very promising and is a feature that many people will seek for the IOT implications. 
Also just to give you some context, I have developed with some friends some indoor sensors (the classical temp humid pressure etc) that we now want to attempt to mesh together and stream data to Azure (EventHub). Not being the firmware guy, I am wondering what kind of restrition your protocol requires when sampling data from I2C device i.e. shall I only read when the device wakes up for the mesh comms or shall I also schedule a regular wake up for sensor readings.
Since you designed it, what is the reccomended approach when doing telemetry applications?

mariusheil, 2015-08-24:
I haven't worked with I2C yet as I was using UART for all my communication. The sampling would depend on your needs (how often does it need to be sampled and how critical is battery performance for you?) When you write a Module for your sensors, you can simply use the provided TimerEventHandler which will be processed right along with everything else. As I haven't worked with external sensors yet, you should probably just give it a try. I'm interested for your feedback :-)

mariusheil, 2015-08-28:
I'll close this question, so it does not turn up under open issues but you can keep on asking :-)

closed 2015-08-28 by mariusheil