bluerange-io/bluerange-mesh #8: Measure advertising packet rssi of surrounding nodes

mariusheil, 2015-08-21:
The rssi of received advertising packets from surrounding nodes should be saved and reported with the connections packet so that a useful distance-graph can be built with this data.

alanzheng800, 2016-03-30:
hi,mariusheil
i use cmd "action 0 status get_nearby" to one node ,but it return 
"{"nodeId":13360,"type":"nearby_nodes","module":30,"nodes":[]}"
i am sure it had connected to another node,why?
ps:i have two node ,they work good .

mariusheil, 2016-04-05:
Nearby nodes doesn't return connected nodes. It does only return nodes that it recognized through advertising packets. (when receiving a JOIN_ME packet). You can instead type action 0 status get_connected to get all connected nodes.

closed 2016-04-08 by mariusheil