bluerange-io/bluerange-mesh #11: Does FruityMesh work on the nRF51 SmartBeacon Kit PC

mariusheil, 2015-08-28:
From tkdom: Curious, does fruity mesh run on the smart beacon?  Nrf51 PCA20006?

mariusheil, 2015-08-28:
I haven't tested it since I do not have access to a smart beacon kit, but I don't think there's a reason which would prevent it from running. Maybe you have to change the LED configurations, but that should be all.

closed 2015-08-28 by mariusheil

dburongarcia, 2015-10-23:
Hi, I was wondering the same. Has anyone tried it before closing this issue? Thanks

mariusheil, 2015-10-26:
Hi, have you checked if the S130 runs on this kit? If not, fruitymesh will not run. If I've read correctly, the beacon kit must be updated over the air but does not support softdevice updates, that would make it hard as well, also it might only have 16kb of ram which doesn't make it easier. do you have a specific use case in mind?

dburongarcia, 2015-11-05:
Hi, sorry for the delay. I have tried these days to program the beacons using a wired connection to a NRF51 DK but it seems that S130 will run just from rev 1.2 of the beacon kits :( However, I've tried with the Waveshare's Core51822 modules and they work great!

dburongarcia, 2015-11-05:
Summary: NRF Beacon kits below 1.2 are not compatible. Be aware of that if you order them, usually the shops don't give that information. I'm having the best results with the QFACA1 chips.

mariusheil, 2015-11-05:
Great to hear :-)