bluerange-io/bluerange-mesh #12: OSX development environment and Makefiles

mariusheil, 2015-08-28:
@tkdom wrote:

Thanks Marius!

Nice work on the mesh!  I am going to try to get your binaries built under OS X with gcc if I can this weekend.  I was hoping there would be makefiles, but I am happy you open sourced the code, so I am not comPlanning :). Unfortunately, I am not that familiar with eclipse.  I have all the nrf51 SDK 9 examples running under the gcc tool chain on OS X.  I would assume the nrf51822 beacon pca20006 would work, but my initial attempt was not successful.  I'll give it another shot after I figure out how to actually build the fruity mesh binary from scratch.

If I can get all that stuff set up, I will use fruity mesh as a basis for an internal technology workshop in October with ThoughtWorks.  Thanks for all the great work.  Any pointers you might have in addition to what is on the wiki for getting the build environment set up would be appreciated.  If I can get make files working with SDK 9 I will certainly send you a pull request...

Tim

mariusheil, 2015-08-28:
Hi,

there are no makefiles because when I started with development, the SDK was not GCC compatible. Have you been able to compile C++ projects with the official makefiles? If yes, that should work with FruityMesh as well.

I've just pushed something to this repository: https://github.com/mwaylabs/fruitymesh-devenv
It's mostly for windows users, but I guess you can learn a lot from the filestructure and modify the setup batch script a little to create symlinks or hardlinks for unix based systems.

Marius

ihassin, 2015-10-19:
I think we can close this now - we have a makefile committed at the root of the project

mariusheil, 2015-10-19:
you're right ;-)

ihassin, 2015-10-27:
This can be closed - makefile works and there's also [this](https://github.com/ihassin/fruitymesh-mac-osx).

closed 2015-10-30 by mariusheil