bluerange-io/bluerange-mesh #16: Can't implement custom module

microcosm, 2015-09-26:
Hi,

I am following the instructions here:
https://github.com/mwaylabs/fruitymesh/wiki/Implementing-a-Custom-Module

When I comp[ile from step (3), I get the following compilation error:
Linking target: FruityMesh.elf
/var/folders/qj/bys33jmj7p32jn6knf3bd9cw0000gn/T//ccBIx6jT.ltrans5.ltrans.o: In function `Node::Node(unsigned short)':
/Users/admin/nrf/projects/fruitymesh/./src/mesh/Node.cpp:107: undefined reference to `PingModule::PingModule(unsigned short, Node*, ConnectionManager*, char const*, unsigned short)'
collect2: error: ld returned 1 exit status
make: *** [all] Error 1

microcosm, 2015-09-26:
btw I have included `PingModule.h` from `Node.cpp` (before you ask)

microcosm, 2015-09-26:
OK, fixed. You have to manually add it as a source to Tim's makefile...
https://github.com/mwaylabs/fruitymesh/blob/master/makefile

closed 2015-09-26 by microcosm

anujdutt9, 2016-09-21:
Hi. I am trying to compile the code but I am not able to do so. Can you please help me out.
Thanks : )