bluerange-io/bluerange-mesh #18: Node has old ID after enrollment

microcosm, 2015-09-28:
Hi, I am attempting to enroll a node with an ID of 1. From the console I get the following output and the status LED goes green as expected. However after a hard reset the node ID is still 13314. Is there something extra I need to do to write-enable the persistent storage? Or can you suggest how to debug this?

```mhTerm: action 13314 enroll nodeid 1 1
[EnrollmentModule.cpp@186 ENROLLMOD]: Enrollment (by nodeId) received nodeId:1, networkid:1
[Storage.cpp@118 STORAGE]: Writing len:34 to block:0
[Storage.cpp@71 STORAGE]: Event: 3, result:0, len:128
[Storage.cpp@71 STORAGE]: Event: 1, result:0, len:34```

mariusheil, 2015-09-28:
Hi,

do you have enabled persistent Storage for the node? Its available In the Config. If not, the values will not be written persistently.

Marius

microcosm, 2015-09-29:
Do you mean these two?
`bool ignorePersistentNodeConfigurationOnBoot = false;`
and:
`bool ignorePersistentModuleConfigurationOnBoot = false;`

I have switched them both to `false`, and I get exactly the same results. LED goes green but the node ID is not changed after reset.

mariusheil, 2015-09-29:
the other way round ;-) use true.

mariusheil, 2015-09-29:
sorry,wrong,... you were right of course....

microcosm, 2015-09-29:
Does this work for you? Are you able to enroll nodes with new IDs?

mariusheil, 2015-09-29:
Yes, that works fine for me. Ill be able to test it in one week when i am back from holidays. i have some code that i have not yet pushed to github, but I dont think there was anything in there. you could listen for the outputs of the Storage class and look what's bring written.

mariusheil, 2015-10-06:
Hello,

how are things working for you? I'm back from  holidays and can now look into the problem. Did you manage to enroll nodes?

closed 2015-10-30 by mariusheil