bluerange-io/bluerange-mesh #19: Data sent across mesh not same as data sent locally

microcosm, 2015-09-29:
Hi,

I have been doing my best to send a string of characters from one node to another using a custom module. The odd thing is it works perfectly when the node sends the message to itself. However when it sends to another node on the mesh only part of the message gets through.

(I can see in the ConnectionManager that when the target nodeID is local it simply fires the relevant local `Node` callback rather than using a network connection, which makes sense. But as far as I can tell, the message data is not altered or corrupted prior to the `sd_mutex_release` call in `PacketQueue::Put`.)

I'll show you. This code is in the `TerminalCommandHandler` of my custom module:
```cpp
connPacketModuleAction packet;
std::vector<u8> myVector(commandArgs[2].begin(), commandArgs[2].end());
for(int i = 0; i < myVector.size(); i++) {
    packet.data[i] = myVector[i];
    logt("GATEWAYMOD", "%d", packet.data[i]);
}
```

And this is in `ConnectionPacketReceivedEventHandler`:
```cpp
connPacketModuleAction* packet = (connPacketModuleAction*)packetHeader;
logt("GATEWAYMOD", "%d %d %d %d", packet->data[0], packet->data[1], packet->data[2], packet->data[3]);

string testString = "";			
for(int i = 0; i < sizeof(packet->data); i++) {
    testString += packet->data[i];
}
logt("GATEWAYMOD", "'%s'", testString.c_str());
```

When I send the message back to the local node ID, here is the output:
```
[GatewayModule.cpp@103 GATEWAYMOD]: 116
[GatewayModule.cpp@103 GATEWAYMOD]: 101
[GatewayModule.cpp@103 GATEWAYMOD]: 115
[GatewayModule.cpp@103 GATEWAYMOD]: 116
[GatewayModule.cpp@119 GATEWAYMOD]: 116 101 115 116
[GatewayModule.cpp@125 GATEWAYMOD]: 'test'
```

However when I send remotely, here is the output from the source node:
```
[GatewayModule.cpp@103 GATEWAYMOD]: 116
[GatewayModule.cpp@103 GATEWAYMOD]: 101
[GatewayModule.cpp@103 GATEWAYMOD]: 115
[GatewayModule.cpp@103 GATEWAYMOD]: 116
```

And here is the output from the target node:
```
[GatewayModule.cpp@119 GATEWAYMOD]: 116 0 0 0
[GatewayModule.cpp@125 GATEWAYMOD]: 't'
```

As you can see, only the first byte in the array is received. I have exactly the same code running on both nodes.

mariusheil, 2015-09-29:
you have probably not entered the right Message size into the sendmessage method, the length is the sum,of the packet Headers plus your data. try to increase the size by one or two to see if that us the Problem, then check in another Module for how you should calculate the size.

microcosm, 2015-09-29:
Awesome! Thankyou. This works:
```cpp
cm->SendMessageToReceiver(NULL, (u8*)&packet, SIZEOF_CONN_PACKET_MODULE_ACTION + testString.length() + 1, true);
```
For some reason I had it in my head that the message sizes were fixed.

closed 2015-09-29 by microcosm