bluerange-io/bluerange-mesh #20: Connect iOS device to FruityMesh

henkejon, 2015-10-06:
Hi FruityMesh,

Is it possible to connect a iOS device to a FruityMesh node/network? Could it be done with nRF/BLE UART or similar?

Regards
Henke

mariusheil, 2015-10-06:
Hi, that should be possible. It is definitely on the roadmap to write a module that allows non-mesh devices to connect to the network, but that does not yet work out of the box. Using a UART over BLE service would definitely be a start. I don't know how much overhead the UART service has, but I don't think it should be much.

henkejon, 2015-10-08:
Hi,
There is a good example of UART over BLE in the SDK:  nrf_sdk\examples\ble_peripheral\ble_app_uart
Thanks! :-)

dburongarcia, 2015-11-21:
Did you manage to get it done? I'm looking for something similar in Android.

dburongarcia, 2015-11-21:
It would be great if we could send commands wirelessly using a smartphone app like the nRF Toolbox, available for both iOS and Android.

mariusheil, 2017-08-09:
being worked on, but may take some time for the update to arrive :-)

closed 2017-08-09 by mariusheil