bluerange-io/bluerange-mesh #29: Serial port pins...

Protoneer, 2015-11-17:
I am using the following board : http://www.waveshare.com/wiki/BLE400

I ran the deploy scripts and it successfully uploaded the softdevice and firmware.

When trying to connect to the serial port it does not respond. Can you tell me what pins are used for the serial?

mariusheil, 2015-11-17:
Hello,

the header file for the nrf51 dongle is in the /config folder. This header file is included in the Config.h file and includes the pin configuration of the board. I am currently on my phone, but you should be able to find the pins easily in this file.

dburongarcia, 2015-11-21:
I'm using the same board and modules and I think that I'm not having any problems. I'll try it when I get to the office and let you know. Have you tried the solution given by mariusheil?

closed 2015-12-01 by mariusheil