bluerange-io/bluerange-mesh #30: Read IO pins

dburongarcia, 2015-11-21:
I wonder if the IO module would be able to read a button/sensor connected to the io pins so it could, for example, use the action module to do an enrollment? Thanks

mariusheil, 2015-11-23:
Hi, it does not read a button or sensor yet, but that's possible without a doubt ;-)

dburongarcia, 2015-11-23:
Is there any timeplan to include this? Just in case, do you have any tips to modify the IO module to get it done?

mariusheil, 2015-11-23:
I don't have a timeframe right now. It's definitely on the list but I have
to get a few other things done before I can implement this. I'm not sure
yet, what to include in the IO module because I don't want too much logic
in there that is specific to one case, but I think a publish and subscribe
mechanism for observing specific buttons would certainly help. Maybe a
function that allows to register a listener for a GPIO pin that can either
trigger a packet to be sent or dispatch an event.
But if you just need one specific button, you could also implement this
logic in your own module and then react upon it.

closed 2015-12-01 by mariusheil