bluerange-io/bluerange-mesh #31: Sorted nodes' IDs

dburongarcia, 2015-11-21:
Is there any way that as soon as a node enter the mesh it checks the rest of the node IDs to sort itself in an ascending order? Thanks

Example: first node is node 1, second is node 2 and so... if node 2 turns off and there are 10 nodes the next one would be node 11 or node 2, depending on the situation.

mariusheil, 2015-12-01:
I'll just answer this one on GitHub as well:
Currently, the clustering algorithm relies on the nodeId's being unique in a network. I've designed it this way because in most use-cases you'll have a mesh that you don't want to be open for anyone, so you can easily control the nodeIds inside. You will also want to know which id is which device. Assigning them with the mesh poses a few problems:
- One could query all existing nodes in the mesh for their nodeIDs which will result in some overhead. Furthermore, what happens to nodes that are not currently connected to the mesh? Will their nodeIDs be reused? And if yes, how could a joining node that was once part of the mesh know that its nodeId had not been used by another node?
If it's a small mesh, it would probably work by generating random nodeIDs. It's not one of the best ideas, though ;-)

dburongarcia, 2015-12-01:
Hi, the idea was just a way of easily identify nodes manually. We are deploying the beacons sequentially and we need to know the 'location' with an easy ID and how many nodes are deployed. Wouldn't it be possible to know on each node activation how many nodes are in the mesh and just assigning that to the id? It's ok if there are 'holes' in the list of IDs, we just need to know the sequence of deployment.

mariusheil, 2015-12-02:
So you are placing them on a virtual map? Could you describe the process a little more?

closed 2016-04-08 by mariusheil