bluerange-io/bluerange-mesh #34: When the 1.0 version will be released?

qianqiangliu, 2016-01-25:
I want to use FruityMesh in my product, so I am really looking forward to the 1.0 version coming soon!

mariusheil, 2016-01-26:
Hi, I am currently doing a lot of stability testing and improvements. During testing I have found that there is an error in the current S130 v1 which is now going to be fixed in the upcoming v2. Another error has occured yesterday that is currently being tracked down. So releasing really depends on how much longer it takes me to put in some critical fixes. It does also depend on your type of project. How many nodes are you planning to use, which connection intervals, which kind of batteries / power source, etc,... Could you provide some details?

qianqiangliu, 2016-01-26:
Thank you for your reply!I want to use FruityMesh in SmartHome area, so it's about 30 ~ 50 nodes I am planning to use

qianqiangliu, 2016-01-26:
Thank you for your reply!
I want to use FruityMesh in SmartHome area, so it's about 30 ~ 50 nodes I am planning to use.
I have some sensor nodes, powered by button cells, and the power must be low enough for one year or longer time. Also, FruityMesh should be easy to configure and operate. So far, you done well, and I hope FruityMesh will be the revolution in SmartHome or IoT field.

mariusheil, 2016-01-26:
Thanks, I am currently also working on improving the meshing for low power because there's some problems with connection timeouts that could be handled more gracefully. But this is going to take some more time and its hard to tell an exact date when these improvements are integrated into the mesh, but it's definitely something I want to have in version 1.0

closed 2016-04-08 by mariusheil