bluerange-io/bluerange-mesh #35: Enumerating GATT Services and Characteristics of a Node

corymast, 2016-02-18:
Hello,

I'm wondering if it's possible to use the FruityMesh algorithm to connect to a particular node in the mesh and enumerate the GATT services and characteristics of that node. For example, if I have a node with a temperature sensor, can I connect to that node and read the temperature from a custom temperature characteristic? In this example application, I would connect a node to a host PC over UART and then use that node to request the temperature of a given node. Once that data is received at the requesting node, it would relay that data over UART to the host PC.

This is a great mesh algorithm and any help you can provide in achieving this would be awesome.

Thanks!

mariusheil, 2016-02-18:
Hello,

this is certainly possible and is on my future roadmap, but currently I did
not have the time to work on this feature but I am working on making the
mesh more stable. I will be on holidays next week but I plan to publish a
new version of Fruitymesh after this week.
There are some steps that are needed to implement this feature. The first
thing to start with would be to modify the Connection so that it supports
different types. I think there's already an enum for different types
floating around somewhere (SINK, non-mesh, mesh). Then, after the
connection is modified it should not count against anything that concerns
meshing and should not do a standard handshake but a service discovery
(which should also still be somewhere in the code).
I think I can add some boilerplate code for this after my holidays together
with the new version.

Marius

corymast, 2016-02-19:
Hi Marius,

Thank you for your quick reply. I will start with your suggestions and work from there to get it up and running. I suppose I could also use the get_config module and pass it a custom parameter, i.e. temp instead of adv, to get started until I can implement the service discovery feature. I look forward to your new version after your vacation. I would love any boilerplate code you come up with.

Enjoy your holiday!

Cory

mariusheil, 2016-02-19:
If the temperature sensor broadcasts its information in the advertising
messages, you could use the scan module. That's the easiest way. To connect
to another non-mesh device, there's probably going to be some lines of code
that want to do meshing with that device.

corymast, 2016-02-20:
Thanks Marius. My other idea was to implement a custom module using the tutorial but I think it might be worth the coding effort to interface the mesh with the sensor device since I would eventually like to be able to add more sensors to some of the nodes and all of the sensor data may not fit in the advertisement data. I will keep you updated on my progress.

closed 2017-08-09 by mariusheil