bluerange-io/bluerange-mesh #37: Adding Modules

junaediadi, 2016-03-02:
Dear Marius,

Today,I have received NRF51dk board and trying fruity mesh. I'm following your tutorial to makae ping module,but when I typed "modules",it says:
"Command not found
{"user_input":"modules"}
"
I also have searched in node.cpp that no command for "modules"
Could you help me to solved this?

Thanks,

Junaedi Adi P
[image]

mariusheil, 2016-03-03:
Hello,

I haven't flashed the old version on one of my beacons, but according to
the wiki the command should be get_modules this or get_modules [nodeID].
https://github.com/mwaylabs/fruitymesh/wiki/Modules
Have you tried that? I just corrected it in the tutorial, hope it works for
you.

junaediadi, 2016-03-03:
Dear Marius,
It works with this command,but it doesnt work when I tried to use "pingmod".
I have tried to activate the module,but it always come back to 0 again.
[image]

junaediadi, 2016-03-03:
and also the ping module always give notation question mark in eclipse.
[image]

mariusheil, 2016-03-03:
Hi,

the question mark is only about the git repository and tells you that this
file has not yet been added to it. Could you post your code somewhere like
http://pastebin.com/ ? (The Pingmodule.cpp and the Node.cpp)
- Check again, what you have in your TerminalCommandListener.
- Did you register the Terminal Command listener in the beginning of the
module?
- Try to use some log outputs to find out which methods are called or debug
the Program.

junaediadi, 2016-03-03:
this is the node.cpp:http://pastebin.com/y26FfFUs and 
this is the ping.cpp"http://pastebin.com/3UaJC5XM.
maybe I miss something from your tutorial,I will check it while I have sent the code to you.

mariusheil, 2016-03-03:
I guess the error is around line 83. Have a look at this:
http://pastebin.com/CTKgH76z

junaediadi, 2016-03-03:
yes you are right,just directly use the inside the command handler.
bool PingModule::TerminalCommandHandler(string commandName, vector<string> commandArgs)
{
		//React on commands, return true if handled, false otherwise
		if(commandName == "pingmod")
........................

at the last implementation you said that we can make a counter using status reporter module.I did not understand the last part.should I make function inside this:

bool StatusReporterModule::TerminalCommandHandler(string commandName, vector<string> commandArgs).

Thanks...

mariusheil, 2016-03-03:
> vector commandArgs).

That part does just say that normally, a ping function would be included in
the StatusReporterModule, but I have not done so for the tutorial. If you
want to measure the time that it takes for the ping to come back, you have
to give a handle to the ping or send it together with the sent-time, then
you can calculate how long it took.

junaediadi, 2016-03-03:
what about if I print the log time together when I send the ping command:for example:

pingmod 45
print the current time..

and in the receiver:
ping command received..
print current time..

is it possible with that?

junaediadi, 2016-03-04:
Dear Marius,
is it possible to change the power of fruity mesh?
for example I want to change to 4 dBm likes using ble_gap_tx_power_set(4);

closed 2016-03-09 by junaediadi