bluerange-io/bluerange-mesh #38: Compile error on Mac OS X

makerdiary, 2016-03-08:
Hi,
I use nRF51_SDK_9.0.0_2e23562 and gcc-arm-none-eabi-5_2-2015q4-20151219-mac and encounter errors as follows:

`Linking target: FruityMesh.elf
./deploy/sdk/nrf_sdk_9_0/components/softdevice/s130/headers/ble_gap.h: In function 'sd_ble_gap_rssi_stop':
./deploy/sdk/nrf_sdk_9_0/components/softdevice/s130/headers/ble_gap.h:1286:1: warning: asm operand 0 probably doesn't match constraints
 SVCALL(SD_BLE_GAP_RSSI_STOP, uint32_t, sd_ble_gap_rssi_stop(uint16_t conn_handle));
 ^
./deploy/sdk/nrf_sdk_9_0/components/softdevice/s130/headers/ble_gap.h:1286:1: error: impossible constraint in 'asm'
./deploy/sdk/nrf_sdk_9_0/components/softdevice/s130/headers/ble_gap.h: In function 'sd_ble_gap_rssi_start':
./deploy/sdk/nrf_sdk_9_0/components/softdevice/s130/headers/ble_gap.h:1272:1: warning: asm operand 0 probably doesn't match constraints
 SVCALL(SD_BLE_GAP_RSSI_START, uint32_t, sd_ble_gap_rssi_start(uint16_t conn_handle, uint8_t threshold_dbm, uint8_t skip_count));
 ^
lto-wrapper: fatal error: /Users/jerrin/workspace/meshlight/fruitymesh/deploy/gcc-arm-none-eabi/bin/arm-none-eabi-g++ returned 1 exit status
compilation terminated.
/Users/jerrin/workspace/meshlight/fruitymesh/deploy/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [all] Error 1`

Have ever encountered these error?
Thanks,
Zelin

junaediadi, 2016-03-09:
i think that is similar with this problem:
https://github.com/mwaylabs/fruitymesh-devenv/issues/2

did you follow the readme file?

makerdiary, 2016-03-10:
You are right. Thank you very much.

closed 2016-03-10 by makerdiary