bluerange-io/bluerange-mesh #39: Test Battery

junaediadi, 2016-03-09:
Dear Marius,

I have question how to use test battery.cpp?
For example I want to do measurement with  those function (void TestBattery::advertiseAt100ms()).
Could you give me an example and how to implement that function?

Sincerely yours,

Junaedi Adi Prasetyo

mariusheil, 2016-04-05:
Hello,

basically, this class should be used instead of the other initialization code in the main() function. It will then only execute some specific actions so that measurements can be done. By calling prepareTesting, some initialization code is executed which will then disable the default Node behaviour of discovery, connecting, etc,....

junaediadi, 2016-04-05:
Dear Marius,
I have done with that.Thanks so much. Below is the result of advertising and scanning:
[image]

[image]

mariusheil, 2016-04-05:
Perfect, looks good. Seems like you us the same scope that I used :-)

junaediadi, 2016-04-05:
I use MSO X 3024 A.
I have question. I use advertising interval and scanning interval 100 ms, the rebuild new connection of two devices still take time around 2-3 seconds.is it right?

how to make the rebuild new connection at fruity mesh faster?

mariusheil, 2016-04-05:
Hi,

it works the following. A node will scan for a longer interval to find a good set of other nodes that it could connect to. It will then decide which of these is the best connection partner. It cannot just connect to the first one available. Therefore it will take some time until it connects. This is controlled through the discovery and back off times.

junaediadi, 2016-04-05:
I understand now. My idea now is to get the best setting for that,for example how long time to make new connection based on different times interval and scanning.
I will share the results of that to you and you can give the input to me.

closed 2016-04-05 by junaediadi