bluerange-io/bluerange-mesh #42: UART communication bug

makerdiary, 2016-03-31:
Hi guys,
I use UART for communication with a control application. nRF51822 will miss some characters when the host sends a string to it. Maybe it has bug in simple_uart. Have you ever meet this problem?
I am trying to use FIFO in uart.

Thanks.

alanzheng800, 2016-04-01:
use fifo_uart will be OK ,i have try it .

makerdiary, 2016-04-02:
Yes, fifo uart is OK.

mariusheil, 2016-04-05:
Do you use hardware flow control? I do sometimes notice problems where a command is not recognized properly. It is probably better to use an interrupt driven UART implementation.

alanzheng800, 2016-04-07:
i don`t use flow control .but i use a new fifo to buffer data ,then use old code to receive it one by one.and must use a counter to count time.if not do that ,it will not work good.

mariusheil, 2016-04-08:
Is this bug still open, or closed? I am not sure?

makerdiary, 2016-04-13:
FIFO should be used.

closed 2016-04-13 by makerdiary