bluerange-io/bluerange-mesh #43: nRF52 support state

Informatic, 2016-04-01:
Hello!
I'm wondering - what is the current state of nRF52 support in FruityMesh? So far I've been able to build it against `nrf52_sdk_0_9_2` after some fixes to the makefile, but it seems to cause HardFault inside of SoftDevice code (1.0.0-alpha attached with that sdk version, just to be sure) right on `softdevice_handler_init` in `bleInit`. I might be missing something since that's my first take on Nordic products. Has anyone been able to run FruityMesh on nRF52? I'm running Preview/Engineering A revision, precisely, I'll try it on recent Development Kit soon.

Informatic, 2016-04-04:
Ok, the problem seemed to be with linkerscript / Makefile / CMSIS, after ripping CMSIS out and rewriting Makefile mostly from scratch everything build fine and works with two nodes, I'll be doing more testing soon.

closed 2016-04-04 by Informatic

ihassin, 2016-04-04:
@piotr: Great news about the makefile. Can you send a link to it? I would like to learn about the differences between 51 and 52.

Thanks!

Informatic, 2016-04-04:
I'll update my fork in a couple of days after settling it down a little bit and will drop a message in this issue if you'd like to.

ihassin, 2016-04-04:
Thanks, appreciated.

Thanks,
Itamar

Informatic, 2016-04-06:
@ihassin Just notifying - I've pushed some of my rage cleanups, including cleaner makefile into my fork here: https://github.com/Informatic/fruitymesh

ihassin, 2016-04-06:
Great thanks!

Thanks,
Itamar