bluerange-io/bluerange-mesh #45: Data Rate

junaediadi, 2016-04-07:
Dear Marius,
What is the data rate/throughput of Fruity Mesh?

I tried to send large data with 80 characters and the chip is hanging(need to reset),but with 70 characters is still okay.

mariusheil, 2016-04-07:
How did you send it? as a module message? Throuput is controlled by connection interval and the SoftDevice used. I am currently preparing a commit with S130 v2 support. Larger packets get split into multiple packets with 20 bytes each. But there is a limit to the size of a packet because it has to be reassembled at the destination node. This buffer size can be set in the config. But you could also split your data into multiple packets yourself.

junaediadi, 2016-04-07:
[image].

I tried by edtiting the script in node.cpp like this:
------------
else if(commandName == "datal")
	{
		const u8 dataLength = atoi(commandArgs[0].c_str());
		u8 _packet[dataLength];
		connPacketHeader* packet = (connPacketHeader*)_packet;
		packet->messageType = MESSAGE_TYPE_DATA_1;
		packet->receiver = 0;
		packet->sender = persistentConfig.nodeId;

		for(u32 i=0; i< dataLength-5; i++){
			_packet[i+5] = i+1;
		}
		cm->SendMessageToReceiver(NULL, _packet, dataLength, true);

the buffer size is to be default in config.h:
#define PACKET_SEND_BUFFER_SIZE 400
#define PACKET_REASSEMBLY_BUFFER_SIZE 200

mariusheil, 2016-04-07:
Tryed it with my local version but setting it to 80 worked for me. Maybe I have fixed something in the meantime. Give me some time to publish the new code, and I guess it should work then.

junaediadi, 2016-04-07:
Dear Marius,
Ok. Can not wait for the latest version.

Thank you.

junaediadi, 2016-04-15:
Dear Marius,
Is it possible to change the data type to send data bigger than now?

void SendMessageOverConnections(Connection* ignoreConnection, u8* data, u16 dataLength, bool reliable);
void SendMessageToReceiver(Connection* originConnection, u8* data, u16 dataLength, bool reliable);
I mean that changing the u8 and u16 to be u32 or u64.

Best Regards,
Junaedi Adi Prasetyo

mariusheil, 2016-04-15:
Hello,

doesn't really help to change the data type, because currently it allows
for 65 kb per packet. The problem is however, that this has be be
reassembled to a single packet on the receiving end. And 65kb is already a
lot more than the RAM of the nRF51 can hold. You have to send multiple
packets to send more than 200 bytes (PACKET_REASSEMBLY_BUFFER_SIZE).

junaediadi, 2016-04-15:
Thanks for your explanation.I understand now.

I have tried to send large data,and the maximum that can be sent is 254 bytes (2kB).
I have increased the buffer size to be 1000,but 254 is the maximum that can be reached for 1 single packet.

[image]

mariusheil, 2016-04-15:
I think this is a limitation of the PacketQueue implementation, it uses a
one-byte field when saving packets to store the length. I thought 256 bytes
would be a viable size for packets on devices with a small sized RAM. Most
packets that are sent do only consist of around 40-60 bytes. Everything
larger is probably sensor generated data and can simply be split into
multiple packets. The automatic package splitting is kind of a convenience
function so that it doesn't have to be done manually for all types of
packets.

junaediadi, 2016-04-15:
Dear Marius,

Thanks.Its clear now.

Sincerely yours,
Junaedi Adi Prasetyo

closed 2016-04-15 by junaediadi