bluerange-io/bluerange-mesh #46: Version 0.3 released

mariusheil, 2016-04-08:
Hello everybody,

I finally managed to put up a new version with lots of changes. Sorry I could not release it in smaller steps but there were some problems holding me back. I know that this will probably break some stuff and this is why I need your help.
Do the makefiles still work for you? I changed some things, but it looks like the generated binary is two times the size of the one that I get when I compile it with my eclipse-generated makefiles. The binary for v0.3 should be about this size:
   text	   data	    bss	    dec	    hex	filename
  68335	    344	   5724	  74403	  122a3	FruityMesh.elf
  68335	    344	   5724	  74403	  122a3	(TOTALS)
It it is bigger, something is wrong with the makefile and I do not really know what it is at the moment.

Thanks

Informatic, 2016-04-08:
I've rewritten makefile for previous version to make it a little more maintainable and now I'm adjusting it to 0.3. I'll look into it when I manage to get UART working on nRF52.

junaediadi, 2016-04-08:
Dear Marius,
Thanks for the update. I have tried to compile it and it works. However, I will try to flash it on my boards on Monday. 

This is the result:
Invoking: Cross ARM GNU Create Flash Image
arm-none-eabi-objcopy -O ihex "FruityMesh.elf"  "FruityMesh.hex"
Invoking: Cross ARM GNU Print Size
arm-none-eabi-size --format=berkeley --totals --radix=10 "FruityMesh.elf"
   text	   data	    bss	    dec	    hex	filename
  68303	    344	   5724	  74371	  12283	FruityMesh.elf
  68303	    344	   5724	  74371	  12283	(TOTALS)
Finished building: FruityMesh.hex
Finished building: FruityMesh.siz
23:51:35 Build Finished (took 42s.875ms)

Thanks.

mariusheil, 2016-04-11:
You got this result with a makefile or did you use the eclipse-way of building it?

junaediadi, 2016-04-11:
I use the eclipse-way to build it. Is it still okay with my compile file?

I have tried it with my boards.I feel that the time to rebuild connection using the new version is rather slower than before, but it is more stable.

Informatic, 2016-04-11:
Could someone upload build log and *.map file somewhere, please? I don't have Eclipse set up right now, and that'd make debugging Makefile much easier.

mariusheil, 2016-04-11:
That's perfectly fine and the way that I use to build it. There's still something wrong with the makefile and I thought that you got different results.

Uploaded the build log here:
http://pastebin.com/X9mpa18V
.map file here:
http://wikisend.com/download/991542/FruityMesh.map

junaediadi, 2016-04-11:
For sure. I have uploaded there.

Informatic, 2016-04-11:
``` 
  text    data     bss     dec     hex filename
  66244     340    5736   72320   11a80 _build/fruitymesh.out
```

Simply copied proper flags over to my rewritten makefile and it works fine. (this is actually nRF52 build, but only difference between that and nRF51 is -mcpu=... and defines)

I'll add some flags/variables to choose between nRF52 and nRF51 and make a pull request later this week.

By the way - is ehal actually used? I ripped it out and project did built fine on v. 0.1 (after replacing linkerscripts with ones from official Nordic examples), and so does right now.

mariusheil, 2016-04-11:
Don't think that EHAL is used, is probably outdated because I set up my project from a tutorial when I started writing it. But the linker script has to be the custom one from me in the /linker directory, because it sets some variables that the code depends on.

Informatic, 2016-04-11:
Yes, i did notice that and i just ported that over to these official linkerscripts too. (a single address, as far as i remember) Anyways, i'll sit down and verify if everything works fine now.

Informatic, 2016-04-11:
Ok, almost everything seems to work fine, except for some configuration loading. When `Config::ignorePersistentModuleConfigurationOnBoot` is false memcpy in DebugModule::ResetToDefaultConfiguration causes hardfault. I'm however not sure whether it's caused by nRF52 being not very tested or my linkerscript / makefile.

closed 2017-08-09 by mariusheil