bluerange-io/bluerange-mesh #49: Configuration loading causes hardfault on nRF52

Informatic, 2016-04-11:
```c
memcpy((u8*)configurationPointer, (u8*)(PAGE_SIZE*254+storageSlot*STORAGE_BLOCK_SIZE), configurationLength);
```
(https://github.com/mwaylabs/fruitymesh/blob/master/src/modules/Module.cpp#L67)

This hack causes HardFault on nRF52. I guess easiest fix would be to just migrate everything properly to NewStorage.

mariusheil, 2016-04-12:
Thank you, will need to implement blocks and page-swapping in NewStorage for this to work. I will have to take a look at the new Nordic storage library as well, but the last time I checked, it didn't really support the features that I wanted.

Informatic, 2016-04-22:
Just a note for now: pstorage seems to be saved at `0x7e000` on nRF52, so changing 254 to 504 fixes that. (`504 * 0x400 == 0x7e000`)

closed 2017-08-09 by mariusheil