bluerange-io/bluerange-mesh #51: Change the timer

junaediadi, 2016-04-18:
Dear Marius,

Could you help me how to change the sleep timer of fruity mesh?
You told in the documentation that you use 500 ms timer enabled,but I don't know how to get this value.

Sincerely yours,

Junaedi Adi Prasetyo

mariusheil, 2016-04-20:
mhh, my mail didn't work, sorry. Timer is configured in Config.h and is called mainTimerTickms I believe, currently set to 200ms.

closed 2016-04-20 by mariusheil

junaediadi, 2016-04-20:
Dear Marius,
At the beginning, I was thinking that TimerTickms is the timer, but I'm not sure because it is written that you do measurement with 500ms. So,why I tried to find variable with 500ms.
Thank you so much.

junaediadi, 2016-04-21:
Dear Marius,
I don't know that it happens only in my device or not. I tried to extend the mesh connection interval to be 1500 ms,and device is error.

u16 meshMinConnectionInterval = MSEC_TO_UNITS(7.5, UNIT_1_25_MS); 
u16 meshMaxConnectionInterval = MSEC_TO_UNITS(1500, UNIT_1_25_MS);

[Main.cpp@225 NODE]: Initializing Softdevice version 0x80
[Main.cpp@241 NODE]: Softdevice Init OK
[Main.cpp@266 ERROR]: Ram base at 0x20002e80

mariusheil, 2016-04-21:
Hello, can not check it currently, but could you try to change both min and
max times at the same time? Because I currently don't support
connectionInterval negotiation.

junaediadi, 2016-04-21:
Dear Marius,
I have tried to check it using the same min and max interval. It works until 1100 ms,but if I made it longer, it doesn't work.

the error at this line:
[Main.cpp@225 NODE]: Initializing Softdevice version 0x80
[Main.cpp@241 NODE]: Softdevice Init OK
[Main.cpp@266 ERROR]: Ram base at 0x20002e80