bluerange-io/bluerange-mesh #55: arm-none-eabi-g++: error: unrecognized command line option '--gc-sections'

makerdiary, 2016-05-27:
Hi guys,
Have you ever encountered this problem?

```
Compiling file: crc16.c
Linking target: FruityMesh.elf
"/Users/jerrin/nrf/sdk/gcc-arm-none-eabi-5_3-2016q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0 -mthumb -Os -fmessage-length=0 -fsigned-char -ffunction-sections -flto -fno-move-loop-invariants -Wextra -g -T./linker/gcc_nrf51_s130_32kb.ld -Xlinker  -Wl,-Map,"_build/FruityMesh.map" --gc-sections --specs=nano.specs _build/AdvertisingController.o _build/GAPController.o _build/GATTController.o _build/ScanController.o _build/Main.o _build/Config.o _build/Connection.o _build/ConnectionManager.o _build/Node.o _build/AdvertisingModule.o _build/DebugModule.o _build/EnrollmentModule.o _build/IoModule.o _build/Module.o _build/ScanningModule.o _build/StatusReporterModule.o _build/TestBattery.o _build/Testing.o _build/LedWrapper.o _build/Logger.o _build/LogTransport.o _build/NewStorage.o _build/PacketQueue.o _build/SimpleBuffer.o _build/SimplePushStack.o _build/SimpleQueue.o _build/Storage.o _build/Terminal.o _build/Utility.o _build/app_timer.o _build/ble_radio_notification.o _build/pstorage.o _build/softdevice_handler.o _build/Vectors_nRF51.o _build/simple_uart.o _build/crc16.o -L./linker -L/Users/jerrin/nrf/sdk/ehal_latest/ARM/Nordic/nRF51/CMSIS/Debug -lCMSIS -o _build/FruityMesh.elf
arm-none-eabi-g++: error: unrecognized command line option '--gc-sections'
make: *** [all] Error 1
```

Informatic, 2016-05-27:
Yes. GCC 5.3 is unsupported for now (even by Nordic themselves, as far as I
checked)

GCC 4.9 works perfectly.

Informatic, 2016-05-27:
Seems like this particular error might be fixed by just replacing
--gc-sections with -Wl,--gc-sections in LDFLAGS, but, again, GCC 5.3 is
unsupported and untested at all.

Feel free to report your findings on 5.3 support, though. :)

Laneville, 2017-02-01:
Was this problem ever solved? Sorry for digging up an old thread.

mariusheil, 2017-02-01:
Have you tried downgrading GCC to the correct version? Its not that a
different version would give you any advantages, I suppose.

Laneville, 2017-02-01:
So I have 4.9.2 on my Mac. But it's giving me those 'unrecognized command line' issues that are mentioned above.
In this thread, @Informatic mentions to just replace the code.... I am building this project in Eclipse and was wondering where I would go to replace the code. Tried replacing it in my makefile but you can't do that, cause it just resets.

mariusheil, 2017-02-01:
The makefile is generated by eclipse, you have to add the commands to the
build settings in eclipse. This is called managed build. But you can also
use t he makefile from the latest release to compile the code. If you have
your compiler installed, you should be able to go in the project folder and
type make in a command line.

closed 2017-08-09 by mariusheil