bluerange-io/bluerange-mesh #56: arm-none-eabi-g++: unrecognized option '-Wl,-Map,_build/FruityMesh.map'

kabez, 2016-08-05:
Hi, I cannot compile, I used the fix for --gc-sections in the other post, however I now get this.

`Compiling file: crc16.c
Linking target: FruityMesh.elf
"/usr/local/gcc-arm-none-eabi-4_9-2015q2/bin/arm-none-eabi-g++" -mcpu=cortex-m0 -mthumb -Os -fmessage-length=0 -fsigned-char -ffunction-sections -flto -fno-move-loop-invariants -Wextra -g -T./linker/gcc_nrf51_s130_32kb.ld -Xlinker  -Wl,-Map,"_build/FruityMesh.map" -Wl,--gc-sections --specs=nano.specs _build/AdvertisingController.o _build/GAPController.o _build/GATTController.o _build/ScanController.o _build/Main.o _build/Config.o _build/Connection.o _build/ConnectionManager.o _build/Node.o _build/AdvertisingModule.o _build/DebugModule.o _build/EnrollmentModule.o _build/IoModule.o _build/Module.o _build/ScanningModule.o _build/StatusReporterModule.o _build/TestBattery.o _build/Testing.o _build/LedWrapper.o _build/Logger.o _build/LogTransport.o _build/NewStorage.o _build/PacketQueue.o _build/SimpleBuffer.o _build/SimplePushStack.o _build/SimpleQueue.o _build/Storage.o _build/Terminal.o _build/Utility.o _build/app_timer.o _build/ble_radio_notification.o _build/pstorage.o _build/softdevice_handler.o _build/Vectors_nRF51.o _build/simple_uart.o _build/crc16.o  -L./linker -L/home/deploy/nrf/sdk/ehal_latest/ARM/Nordic/nRF51/CMSIS/Debug -lCMSIS -o _build/FruityMesh.elf
/usr/local/gcc-arm-none-eabi-4_9-2015q2/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: unrecognized option '-Wl,-Map,_build/FruityMesh.map'
/usr/local/gcc-arm-none-eabi-4_9-2015q2/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: use the --help option for usage information
collect2: error: ld returned 1 exit status
make: *** [all] Error 1'

mariusheil, 2016-08-05:
have you tried removing these options? I think that you are using a different compiler version.

kabez, 2016-08-07:
yeah, fixed it by changing it to -M=_build/FruityMesh.map. Not sure if that was right. 
Still can't compile, more errors now, something to do with the SoftDevice error handler. 

```
Compiling file: crc16.c
Linking target: FruityMesh.elf
"/usr/local/gcc-arm-none-eabi-4_9-2015q2/bin/arm-none-eabi-g++" -mcpu=cortex-m0 -mthumb -Os -fmessage-length=0 -fsigned-char -ffunction-sections -flto -fno-move-loop-invariants -Wextra -g -T./linker/gcc_nrf51_s130_32kb.ld -Xlinker  -M="_build/FruityMesh.map" --specs=nano.specs _build/AdvertisingController.o _build/GAPController.o _build/GATTController.o _build/ScanController.o _build/Main.o _build/Config.o _build/Connection.o _build/ConnectionManager.o _build/Node.o _build/AdvertisingModule.o _build/DebugModule.o _build/EnrollmentModule.o _build/IoModule.o _build/Module.o _build/ScanningModule.o _build/StatusReporterModule.o _build/TestBattery.o _build/Testing.o _build/LedWrapper.o _build/Logger.o _build/LogTransport.o _build/NewStorage.o _build/PacketQueue.o _build/SimpleBuffer.o _build/SimplePushStack.o _build/SimpleQueue.o _build/Storage.o _build/Terminal.o _build/Utility.o _build/app_timer.o _build/ble_radio_notification.o _build/pstorage.o _build/softdevice_handler.o _build/Vectors_nRF51.o _build/simple_uart.o _build/crc16.o  -L./linker -L/home/deploy/nrf/sdk/ehal_latest/ARM/Nordic/nRF51/CMSIS/Debug -lCMSIS -o _build/FruityMesh.elf
/usr/local/gcc-arm-none-eabi-4_9-2015q2/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: _build/FruityMesh.elf section `.text' will not fit in region `FLASH'
/usr/local/gcc-arm-none-eabi-4_9-2015q2/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 105220 bytes
/tmp/ccPzSvz3.ltrans8.ltrans.o: In function `SWI2_IRQHandler':
/home/deploy/nrf/sdk/nrf51_sdk_latest/components/softdevice/common/softdevice_handler/softdevice_handler.c:358: undefined reference to `app_error_handler_bare'
/tmp/ccPzSvz3.ltrans8.ltrans.o: In function `intern_softdevice_events_execute':
/home/deploy/nrf/sdk/nrf51_sdk_latest/components/softdevice/common/softdevice_handler/softdevice_handler.c:147: undefined reference to `app_error_handler_bare'
/home/deploy/nrf/sdk/nrf51_sdk_latest/components/softdevice/common/softdevice_handler/softdevice_handler.c:178: undefined reference to `app_error_handler_bare'
/tmp/ccPzSvz3.ltrans8.ltrans.o: In function `StatusReporterModule::SendDeviceInfo(unsigned short, unsigned char) [clone .constprop.9]':
/home/deploy/nrf/projects/fruitymesh/./src/modules/StatusReporterModule.cpp:171: undefined reference to `app_error_handler_bare'
/tmp/ccPzSvz3.ltrans3.ltrans.o: In function `GAPController::startEncryptingConnection(unsigned short)':
/home/deploy/nrf/projects/fruitymesh/./src/base/GAPController.cpp:283: undefined reference to `app_error_handler_bare'
/tmp/ccPzSvz3.ltrans3.ltrans.o:/home/deploy/nrf/projects/fruitymesh/./src/base/GAPController.cpp:69: more undefined references to `app_error_handler_bare' follow
collect2: error: ld returned 1 exit status
make: *** [all] Error 1

```

I've tried replacing the SDK and that didn't work.

Bardo91, 2017-04-27:
Were you able to fix it? I am having the same trouble

zrecore, 2017-04-28:
Perhaps its a linker script issue (flash storage allocation)? Note how the
stack trace complains about .text not fitting

L/home/deploy/nrf/sdk/ehal_latest/ARM/Nordic/nRF51/CMSIS/Debug -lCMSIS
-o _build/FruityMesh.elf
/usr/local/gcc-arm-none-eabi-4_9-2015q2/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld:
_build/FruityMesh.elf section `.text' will not fit in region `FLASH'
/usr/local/gcc-arm-none-eabi-4_9-2015q2/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld:
region `FLASH' overflowed by 105220 bytes
/tmp/ccPzSvz3.ltrans8.ltrans.o: In function `SWI2_IRQHandler':
/home/deploy/nrf/sdk/nrf51_sdk_latest/components/softdevice/common/softdevice_handler/softdevice_handler.c:358:
undefined reference to `app_error_handler_bare'
/tmp/ccPzSvz3.ltrans8.ltrans.o: In function

Bardo91, 2017-04-28:
Hi I ended up changing the routes defined in the makefile with custom locations in my computer and it worked. I don't know exactly were it was but it is related with using the wrong files and/or not finding them. I hope it helps to the following one. If I have time I will try to see if I find exactly where/when the error happens.

closed 2017-08-09 by mariusheil

jiapei100, 2018-05-26:
@zrecore @Bardo91 

Hi, I've got exactly the same issue:

```
23:47:47 **** Build of configuration Debug for project F767ZI_OV5640 ****
make all 
Building target: F767ZI_OV5640.elf
Invoking: GNU ARM Cross C++ Linker
arm-none-eabi-g++ -mcpu=cortex-m7 -mthumb -mfloat-abi=soft -Og -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fno-move-loop-invariants -Wall -Wextra  -g3 -T mem.ld -T libs.ld -T sections.ld -nostartfiles -Xlinker --gc-sections -L"../ldscripts" -Wl,-Map,"F767ZI_OV5640.map" --specs=nano.specs -o "F767ZI_OV5640.elf"  ./system/src/stm32f7-hal/stm32f7xx_hal.o ./system/src/stm32f7-hal/stm32f7xx_hal_cortex.o ./system/src/stm32f7-hal/stm32f7xx_hal_crc_ex.o ./system/src/stm32f7-hal/stm32f7xx_hal_dcmi.o ./system/src/stm32f7-hal/stm32f7xx_hal_dfsdm.o ./system/src/stm32f7-hal/stm32f7xx_hal_dma.o ./system/src/stm32f7-hal/stm32f7xx_hal_dma_ex.o ./system/src/stm32f7-hal/stm32f7xx_hal_flash.o ./system/src/stm32f7-hal/stm32f7xx_hal_gpio.o ./system/src/stm32f7-hal/stm32f7xx_hal_iwdg.o ./system/src/stm32f7-hal/stm32f7xx_hal_jpeg.o ./system/src/stm32f7-hal/stm32f7xx_hal_ltdc.o ./system/src/stm32f7-hal/stm32f7xx_hal_mdios.o ./system/src/stm32f7-hal/stm32f7xx_hal_pwr.o ./system/src/stm32f7-hal/stm32f7xx_hal_pwr_ex.o ./system/src/stm32f7-hal/stm32f7xx_hal_rcc.o ./system/src/stm32f7-hal/stm32f7xx_hal_rcc_ex.o ./system/src/stm32f7-hal/stm32f7xx_hal_sdram.o ./system/src/stm32f7-hal/stm32f7xx_hal_smartcard_ex.o ./system/src/stm32f7-hal/stm32f7xx_hal_sram.o ./system/src/stm32f7-hal/stm32f7xx_hal_tim.o ./system/src/stm32f7-hal/stm32f7xx_hal_tim_ex.o ./system/src/stm32f7-hal/stm32f7xx_hal_uart.o ./system/src/stm32f7-hal/stm32f7xx_ll_fmc.o  ./system/src/newlib/_cxx.o ./system/src/newlib/_exit.o ./system/src/newlib/_sbrk.o ./system/src/newlib/_startup.o ./system/src/newlib/_syscalls.o ./system/src/newlib/assert.o  ./system/src/diag/Trace.o ./system/src/diag/trace_impl.o  ./system/src/cortexm/_initialize_hardware.o ./system/src/cortexm/_reset_hardware.o ./system/src/cortexm/exception_handlers.o  ./system/src/cmsis/system_stm32f7xx.o ./system/src/cmsis/vectors_stm32f767xx.o  ./src/USMART/usmart.o ./src/USMART/usmart_config.o ./src/USMART/usmart_str.o  ./src/SYSTEM/usart/usart.o  ./src/SYSTEM/sys/sys.o  ./src/SYSTEM/delay/delay.o  ./src/HARDWARE/USART2/usart2.o  ./src/HARDWARE/TIMER/timer.o  ./src/HARDWARE/SDRAM/sdram.o  ./src/HARDWARE/PCF8574/pcf8574.o  ./src/HARDWARE/OV5640/ov5640.o ./src/HARDWARE/OV5640/sccb.o  ./src/HARDWARE/LED/led.o  ./src/HARDWARE/LCD/lcd.o ./src/HARDWARE/LCD/ltdc.o  ./src/HARDWARE/KEY/key.o  ./src/HARDWARE/IIC/myiic.o  ./src/HARDWARE/DCMI/dcmi.o  ./src/_initialize_hardware.o ./src/_write.o ./src/main.o ./src/stm32f7xx_hal_msp.o   
/opt/GCCToolChains/gnu-mcu-eclipse/arm-none-eabi-gcc/7.2.1-1.1-20180401-0515/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: F767ZI_OV5640.elf section `.bss' will not fit in region `RAM'
makefile:76: recipe for target 'F767ZI_OV5640.elf' failed
/opt/GCCToolChains/gnu-mcu-eclipse/arm-none-eabi-gcc/7.2.1-1.1-20180401-0515/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 127372180 bytes
collect2: error: ld returned 1 exit status
make: *** [F767ZI_OV5640.elf] Error 1

23:47:48 Build Finished (took 835ms)
```

My makefile looks like [https://pastebin.com/iuxa1nbg](https://pastebin.com/iuxa1nbg) . Can anybody please give me a hand?

Thank you very much...
Pei

mariusheil, 2018-05-28:
What are you trying to do? Your makefile includes a lot of stm components and your RAM reagion is 120 MB too big. Try to remove most of the compilation files first and add it one by one to see what's causing the issue. Besides, does your project have anything to do with fruitymesh? Looks like you are trying to do something else.