bluerange-io/bluerange-mesh #57: Not able to Use commands on Terminal

anujdutt9, 2016-08-15:
Hi, I am new to FruityMesh. I like this idea and have got all the hardware required with the software. The modules paired up pretty nicely and all handshake was good but as I type a command in the terminal in front of  "mhTerm" it dosen't works most of the times. Like when I type "startterm"  then nothing happens. So, please if you can guide me on this like what should I expect to see on using these commands or how to enter these commands. I am pretty sure I am missing something as rest everything looks good.
Please if you can help.
Thanks

mariusheil, 2016-08-15:
Hi, what Terminal software are you using? I think that it might be because of settings. You need to have hardware flow control (RTS/CTS) enabled and the terminal should send \r as an end of line character. Could you try to use Putty? This works really well for me.

anujdutt9, 2016-08-15:
Hi. Thanks for the reply. I am using Putty only with the recommended settings in the Wiki page.
The thing is as I type the "startterm" command, nothing happens. If I try to type something, it terminates and goes back to "mhTerm:". Can you please tell what to expect from this command and if  possible can you share a screenshot of what happens with this particular command.
Thanks : )

mariusheil, 2016-08-15:
can you try a different command? type status for example and see if it outputs anything. If not, type it again. You don't have to use the startterm command as the terminal seems to be enabled already, just try the other commands :-) Before using startterm, you should use stopterm which will enable the mode for machine-to-machine communication, but I wouldn't try these right now. Just test the other commands. Why did you use an older version? it's best if you use the latest version.

anujdutt9, 2016-08-15:
Hi. The "0.3" version is the latest one as I see it, the older one is "1.9" something in binary folder. 
I have used all other commands and they seem to be working fine. Now I want to try the "leds on" & "leds off" as well as "startterm" commands. They don't seem to be working for me as of now.

anujdutt9, 2016-08-15:
I tried using "stopterm" first but then the terminal starts showing up 
mhTerm: stopterm
{"module":0, "type":"error","code":0,"text":"OK"}
{"module":0, "type":"error","code":0,"text" : "Command not Found"}
and so on. 
Can you please guide me on whats the correct way to do this and what should be the output.
Thanks : )

mariusheil, 2016-08-15:
Hi, all code files are available, getting the build to work takes some time though, will improve on this. You should probably skip the stopterm command ;-) action 0 io led on should switch all leds on action 0 io led off should switch them off. I think i killed the leds on and leds off commands some time ago in favour of the new module commands.

anujdutt9, 2016-08-15:
Oh okey, thanks for the update on this : ) So what are the commands for led on and led off ??
action 0 io led ??  or something else. 
Sorry, but I am unable to understand : )

mariusheil, 2016-08-15:
'action 0 io led on'

anujdutt9, 2016-08-15:
Thanks a lot for helping me out : )
One last thing. If I need to send a data "text" from one node to other, how can I do that ??
I tried using "data [sink | hop | broad] " but nothing happened.
Also I tried using  "data [nodeId]" to connect to node but how do I confirm that its connected only to that like for ex. I connect to node:2240 and send a text to that node only, either directly or using other nodes in between as a link to just bounce the message.
Thanks : )

mariusheil, 2016-08-15:
hi, I don't think that I implemented sending strings anywhere because the nodes themselves are not really interested in text strings and are more comfortable with binary actions that they can execute. You should have a look at the sourcecode and the wiki for finding out how to implement sending between nodes. The "data" command will send an action to all connected nodes and they will display a log output.  action NODEID MODULE_NAME COMMAND_NAME will execute an action on the specified node. It will take care of the delivery through the mesh.

anujdutt9, 2016-08-15:
Thanks a lot for this information : ) I'll surely check into that.
How do you advertise a message ?? I see the Advertising Module under the modules and the configurations for that, but how do I actually use it ??
Like is there a command to advertise the data or is it a completely different setup ?

closed 2016-08-17 by anujdutt9