bluerange-io/bluerange-mesh #60: connect to non mesh node

Molinatoxx, 2016-08-26:
Hello,
fruitymesh works really well but my problem is that I would like to get a mesh node to connect to a smartband and then mesh the data to a sink (perhaps also outside the mesh)..is there any way to do it?
Thanks

Informatic, 2016-08-26:
This is not supported OOTB as far as i know, but you should be easily able
to do this in your own module. (Likely after some tiny patches to the mesh
connection management core, as it was not designed with such usecase in
mind)

I think you should look around documentation of Nordic SDK and play with it
a little bit before starting, though.

Molinatoxx, 2016-08-26:
Thanks,

I admit I have no idea where to begin as I am a real beginner here, thanks though.

mariusheil, 2017-08-09:
I am currently in the progress of implementing this, will take some time until its on github, but it's being worked on :-)

closed 2017-08-09 by mariusheil

PriyeshNema, 2021-08-31:
Hi.. I'm also looking for something similar. Is this now implemented in FruityMesh?

mariusheil, 2021-08-31:
Hello,

yes, this is implemented since a long time. Take a look at the MeshAccessModule which allows you to build connections between different nodes and also from a Smartphone to mesh nodes.

Marius