bluerange-io/bluerange-mesh #61: Working of the "Connection Score" Function ?

anujdutt9, 2016-09-15:
Hi. I read your all documents and the code. I didn't quiet get the working of the "Connection Score" Function to make the connection with the other nodes. Can you please explain this or point me to a piece of code that does that. I couldn't find one.
Thanks : )

mariusheil, 2016-09-16:
Hi, maybe you mean the function that evaluates to which node it should connect to? I think I called it ClusterScoreFunction. Basically this function will take all the collected JOIN_ME packets to determine a good connection partner. In the github implementation, this function will use the number of freeIn and freeOut connections. In the C code, you'll find: DetermineBestClusterAvailable which uses CalculateClusterScoreAsSlave and CalculateClusterScoreAsMaster. What the functions do is commented a lot in the source code, so you'll easily understand it. Basically these functions allow you to specify different criteria such as using the batteryPower available. This would lead to a network that forms with the nodes that have the most battery power first, and afterwards the nodes which have the least power available. Because the latter are added as leave nodes to the network, they will use less energy.

anujdutt9, 2016-09-16:
Hi Marius, 
Thanks for the reply. Yes I got that thing cleared in the code.
One more thing, once I send the data from one node to another, if both nodes are connected to terminal, I get to see that data has been received but when I don't have the second terminal, how do I get the acknowledgement for that the data sent has been received apart from the <> that appears on the receiver side every time some command is executed.

Thanks : )

mariusheil, 2016-09-16:
Hey, most commands don't send an acknowledgement because I don't consider
them to be important and it will generate network overhead. If you need to
get an acknowledgement from a command, this has to be implemented. You can
check the IO Module for example. The "led on" action will generate an
acknowledgement from the receiving node. This is implemented with handles.
So you can send action 0 io led on 7 where 7 is the handle. Then you 'll
get a led_on_response with handle 7 and an error code. Use the appropriate
message types. There is e.g. one that is used for sending general messages
which can be generated on one node for example and there are trigger action
and action response. These should be used for your use-case.

closed 2016-09-29 by anujdutt9