bluerange-io/bluerange-mesh #62: Build Error

juttTechnology, 2016-09-19:
Hello,
        i am very interested to run this project. but i am facing problem in the build.
        i follow the guide line can you please confirm this issue.
        
[image]

  some files you are calling the nrf.h file but i didn't found this file in the project. and also sdk_common.h

can you please guide me to fix this issue. please also check the image for more detail.

thank you

juttTechnology, 2016-09-19:
i search little bit then i found the soldtion...means i add the path.
after that i got very strange error.please check the image.

[image]

mariusheil, 2016-09-20:
Hi, I guess that this could come from one of the Problems in the SDK. Have you checked this step? https://github.com/mwaylabs/fruitymesh-devenv#fix-some-of-the-sdk-files. Do all sd_... functions fail for you? Do you have SDK11 or did you use the newer SDK12? I haven't transitioned the project to the new 12er SDK yet, so you should use version 11.

juttTechnology, 2016-09-20:
Hi, Yes you are Right. this error comes from SDK. i am using the version 11.
yes i also did these fix..
besically i did only two things.
1- changes (#define GCC_CAST_CPP (uint16_t) \)   in nrf_svc.h
2- download the ble_radio_notification files from the link post and just replace.

upto now only sd_flash_page_erase function create the problem.
this problem is very strange. i will try to download sdk v11 again and will replace the files. let see if it is usefull.
if you have any suggestion then please

mariusheil, 2016-09-20:
Could you check in the sdk\components\sfotdevice\s130\headers\nrf_soc.h file if the funciton is defined?

I have this in Line 813:
SVCALL(SD_FLASH_PAGE_ERASE, uint32_t, sd_flash_page_erase(uint32_t page_number));

juttTechnology, 2016-09-20:
i just clean and rebuilt then problem is solved...let see my plan is to implement this into the visualGDB.
can you please give me advice to debug this code then i will understand and  i will try to implement into the VisualGDB.

thank you for your quick support.

mariusheil, 2016-09-20:
Cool :-) Debugging the code is not easy I must tell you, because it uses many calls to the Nordic Softdevice. all these calls will freeze the system while debugging because debugging destroys the realtime characteristics. I haven't tried VisualGDB yet because of the licence costs. Developing with Eclipse was just fine for this project. Ant the integrated debugger works wel if there are no softdevice calls. I'd suggest you stay with logging mostly, which is implemented widely in my system. You can enable and disable the different logtags to get a view on what its doing. Try "debug all" for starters, which will show you all that's currently being logged. (be sure to enable logging in the config.)

anujdutt9, 2016-09-21:
@juttTechnology  Hi. I am also trying to compile this but I am getting some errors. Please if you could share what you did from the beginning so that I can figure out where I am wrong or maybe send your .project file.
Also, I have followed all the steps as mentioned in the tutorial but I might be getting wrong at importing all files of the project into Eclipse. Please if you could guide me on how you imported all the files into Eclipse. That'll be a great help. 
Thanks : )

juttTechnology, 2016-09-21:
Hi 
please follow the full guide line and also add all the reference file.if still the error exist then please go to 
Project->c/c++ indexer->Rebuit
i am sure if you will follow the tutorial and add all reference and at the end follow this command then it  will work.
if you still face problem then please show me the image of the error then i will try my best to help you.

anujdutt9, 2016-09-21:
@juttTechnology  Hi. I have followed all the steps but when I try to run the eclipse that comes with the setup, it shows an error as follows. Please if you can help me with this. The one I tried earlier, I use my own version of eclipse that works on my PC. Thanks : )
[image]

mariusheil, 2016-09-22:
Could you have a look in the logfile mentioned?, maybe you'll find it in the error view or in a console. Seems more like an eclipse error than a build error.

juttTechnology, 2016-09-22:
please try to run using command line 
eclipse.ese -clean
this should work..

anujdutt9, 2016-09-22:
I have tried almost everything. even ran it using cmd. But I get this error:

!SESSION 2016-09-22 07:38:04.020 -----------------------------------------------
eclipse.buildId=4.5.0.I20150528-0700
java.version=1.7.0_79
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_US
Framework arguments:  -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -product org.eclipse.epp.package.cpp.product -clean

!ENTRY org.eclipse.osgi 4 0 2016-09-22 07:38:10.164
!MESSAGE An error occurred while automatically activating bundle org.eclipse.equinox.registry (229).
!STACK 0
org.osgi.framework.BundleException: Exception in org.eclipse.core.internal.registry.osgi.Activator.start() of bundle org.eclipse.equinox.registry.
	at org.eclipse.osgi.internal.framework.BundleContextImpl.startActivator(BundleContextImpl.java:792)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.start(BundleContextImpl.java:721)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.startWorker0(EquinoxBundle.java:941)
	at org.eclipse.osgi.internal.framework.EquinoxBundle$EquinoxModule.startWorker(EquinoxBundle.java:318)
	at org.eclipse.osgi.container.Module.doStart(Module.java:571)
	at org.eclipse.osgi.container.Module.start(Module.java:439)
	at org.eclipse.osgi.framework.util.SecureAction.start(SecureAction.java:454)
	at org.eclipse.osgi.internal.hooks.EclipseLazyStarter.postFindLocalClass(EclipseLazyStarter.java:107)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClass(ClasspathManager.java:531)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.findLocalClass(ModuleClassLoader.java:324)
	at org.eclipse.osgi.internal.loader.BundleLoader.findLocalClass(BundleLoader.java:327)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:402)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:352)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:344)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:160)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at org.eclipse.core.internal.content.Activator.start(Activator.java:46)
	at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:771)
	at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:1)
	at java.security.AccessController.doPrivileged(Native Method)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.startActivator(BundleContextImpl.java:764)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.start(BundleContextImpl.java:721)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.startWorker0(EquinoxBundle.java:941)
	at org.eclipse.osgi.internal.framework.EquinoxBundle$EquinoxModule.startWorker(EquinoxBundle.java:318)
	at org.eclipse.osgi.container.Module.doStart(Module.java:571)
	at org.eclipse.osgi.container.Module.start(Module.java:439)
	at org.eclipse.osgi.framework.util.SecureAction.start(SecureAction.java:454)
	at org.eclipse.osgi.internal.hooks.EclipseLazyStarter.postFindLocalClass(EclipseLazyStarter.java:107)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClass(ClasspathManager.java:531)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.findLocalClass(ModuleClassLoader.java:324)
	at org.eclipse.osgi.internal.loader.BundleLoader.findLocalClass(BundleLoader.java:327)
	at org.eclipse.osgi.internal.loader.sources.SingleSourcePackage.loadClass(SingleSourcePackage.java:36)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:398)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:352)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:344)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:160)
	at java.lang.ClassLoader.loadClass(Unknown Source)
	at org.eclipse.core.internal.runtime.InternalPlatform.openOSGiTrackers(InternalPlatform.java:781)
	at org.eclipse.core.internal.runtime.InternalPlatform.start(InternalPlatform.java:703)
	at org.eclipse.core.internal.runtime.PlatformActivator.start(PlatformActivator.java:32)
	at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:771)
	at org.eclipse.osgi.internal.framework.BundleContextImpl$3.run(BundleContextImpl.java:1)
	at java.security.AccessController.doPrivileged(Native Method)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.startActivator(BundleContextImpl.java:764)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.start(BundleContextImpl.java:721)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.startWorker0(EquinoxBundle.java:941)
	at org.eclipse.osgi.internal.framework.EquinoxBundle$EquinoxModule.startWorker(EquinoxBundle.java:318)
	at org.eclipse.osgi.container.Module.doStart(Module.java:571)
	at org.eclipse.osgi.container.Module.start(Module.java:439)
	at org.eclipse.osgi.framework.util.SecureAction.start(SecureAction.java:454)
	at org.eclipse.osgi.internal.hooks.EclipseLazyStarter.postFindLocalClass(EclipseLazyStarter.java:107)
	at org.eclipse.osgi.internal.loader.classpath.ClasspathManager.findLocalClass(ClasspathManager.java:531)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.findLocalClass(ModuleClassLoader.java:324)
[... cut ...]
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:669)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:608)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1515)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1488)

I have been on this for past 2 days, I really need to make it work.
Please help.
Thanks :)

anujdutt9, 2016-09-22:
!SESSION 2016-09-22 16:56:00.002 -----------------------------------------------
eclipse.buildId=4.5.0.I20150528-0700
java.version=1.8.0_101
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_US
Framework arguments:  -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -product org.eclipse.epp.package.cpp.product
I am getting this in the log file. In between, eclipse started once but after that when I tried to open it again it shows this error in log file.

!ENTRY org.eclipse.osgi 4 0 2016-09-22 16:56:00.652
!MESSAGE An error occurred while automatically activating bundle org.eclipse.equinox.registry (229).
!STACK 0
org.osgi.framework.BundleException: Exception in org.eclipse.core.internal.registry.osgi.Activator.start() of bundle org.eclipse.equinox.registry.
	at org.eclipse.osgi.internal.framework.BundleContextImpl.startActivator(BundleContextImpl.java:792)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.start(BundleContextImpl.java:721)
	at org.eclipse.osgi.internal.framework.EquinoxBundle.startWorker0(EquinoxBundle.java:941)

Please if you can help.
Thanks : )

juttTechnology, 2016-09-22:
if he start one time means you have Java installed.i thought may be you have Java prolem.anway please delete the file from the workspace folder and try to run Again.

anujdutt9, 2016-09-23:
@juttTechnology @marius Hey, Thanks for all your help : ) I got the eclipse working somehow and imported the project to it. It looks something like this. I just want to verify, is this correct or do I need to import some other files as well. 
[image]

Secondly, the DFU Module here is showing a "!" . I think this has some error. Where is this folder and how can I rectify this error.
Also, when I click on Run currently, it gives an error as "Launch Failed. Binary not found"

[image]

Thanks : )

mariusheil, 2016-09-23:
can you try to just build it? Does it build? The dfu Module is not part of the release, so this is normal. Please read some Info on the nordic website on how to build for the nrf51 chipset, This will give you a starting point.

anujdutt9, 2016-09-23:
@mariusheil  Hi Marius. I built the code. Can you please see if this is the expected way in which it should build.
[image]

Regards
Anuj

juttTechnology, 2016-09-23:
First try to find the GNU ARM Eclipse in the c:\Program FIle 
mine is look like this 
C:\Program Files\GNU ARM Eclipse\Build Tools\2.6-201507152002\bin

also try to check in the Programm file (86) folder. if you found nothing then please  download and install 
after that try to copy the link like me.

then come into the project->Properties->c\C++ Built -> Envirment->Path 
add this link also in the path variable using the edit.

i hope this will solve your problem.

anujdutt9, 2016-09-23:
@juttTechnology  Thanks a lot for guiding me. But now When I am building the code it shows the error as below:
[image]

@juttTechnology Can you please have a look at this image and tell if any of the modules are missing or if I need to add any more modules to this folder. Also, I have not been able to find this "sdk_common.h" file anywhere. 
Sorry for bothering you too much but I need this to be working. I have been on this for past 3 days.
Thanks : ) :+1:

juttTechnology, 2016-09-23:
please try 

Project->c/c++ Index-> refresh all files

anujdutt9, 2016-09-23:
Tried that, still this "sdk_common.h" error is there. It says no such file or directory.

juttTechnology, 2016-09-24:
you can canot make the folder by your own..
you have to install the GNU ARM Eclipse.

anujdutt9, 2016-09-26:
@juttTechnology @mariusheil  Hi. I got almost everything working, just this one small error. If you can please help me in this.

[image]

Thanks : )

anujdutt9, 2016-09-26:
@juttTechnology @mariusheil Thanks for all your help. It works. Thanks : ) :+1:

closed 2016-11-09 by juttTechnology