bluerange-io/bluerange-mesh #64: issue in the variable  (__application_ram_start_address)

juttTechnology, 2016-09-28:
Hello,
        i am working with VisualGDB. i manage all the code now i have one one error, which belong to this variable  **__application_ram_start_address**

[image]

 here is the problem in the **main.cpp** file. the undefined reference. when i checked it was in the config file with **extern**. 
it is set in the **linker Script**. can you please explain me this litte bit.because  i really have no idea how i can set this here.

thank you

mariusheil, 2016-09-29:
Hi,

can you try to set the variable to this manually?: 0x20002e80
Make sure to enable logging and it will tell you if an error occured
I'd have to read into the specifics of this again. I currently forgot. This has something to do with the ble_enable params.

Informatic, 2016-09-29:
Yes, this variable is filled by linkerscript and it works perfectly when
building. Haven't really worked with visualgdb myself, generic gdb won't
oppose to that (because... Well, it doesn't do code introspection this deep
afair...)

juttTechnology, 2016-10-04:
hi,, Can you please check this line of code.
this is from Module.cpp Line nr.55   
my software is fully block there.i think some issue from the source or destination memory  size 

memcpy((u8*)configurationPointer, (u8*)(PAGE_SIZE*254+storageSlot*STORAGE_BLOCK_SIZE), configurationLength);

Here is my disassembly View 
[image]

juttTechnology, 2016-10-04:
i follow the other thread ..and i write 126 ...it worked...

juttTechnology, 2016-10-04:
almost the code is working ...i have one issue in the NRF_ERROR_DATA_SIZE 
this error is raised from the  sd_ble_evt_get, which is located in the while loop.
interesting thing is its working for random number of times and after that he give me the error of NRF_ERROR_DATA_SIZE.

mariusheil, 2016-10-05:
Probably something compiler specific. Check if the buffer - into which the event data is copied - is correctly aligned to 32 bit and is big enough. You can read in the nrf devzone as well about this problem.

juttTechnology, 2016-10-07:
Hi 
   Eclipse is working perfectly with nrf51. but not nrf52 
 
here are the changes that i made 
1-Change the Name in the Preprocessor for Nrf52 
[image]

then i also change the path in includes for ehal, from 51 to 52
[image]

then i try to build and i got this error 
[image]

here i mention in the Red box
"C:/nrf/projects/fruitymesh/linker/gcc_nrf51_s130_32kb.ld""

when i am using Nrf52 then why he is taking gcc_nrf51_s130, where i can adjust this issue.
i thought these error comes because of this.

also 2nd error 
C:\Users\ATIF~1.SHA\AppData\Local\Temp\ccWzvHBJ.s:2082: Error: instruction not supported in Thumb16 mode -- `subs r0,r0,#1'

can you please guide me to fix these issue.
First of all i need to run nrf52 on Eclipse then i will move on Visualgdb.

thank you

mariusheil, 2016-10-07:
Have a look under c/C++ Build => Settings => Settings (Folder icon) => Cross ARM C++ Linker, it should be somewhere there where the linker script is configured

juttTechnology, 2016-10-07:
good.that one is fixed.
but the 2nd error is still there
this is something look like assembler
[image]

mariusheil, 2016-10-07:
Did you do a clean on everything? The error itself doesn't tell me anything. Apparently the compiler generated code that the linker is not able to process for the target processor. maybe try to comment everything out except the main class and then try again what parts of the code can be linked without issues?

juttTechnology, 2016-10-08:
Yes i clean all.
i comment all after that the error was still there .
i search alot and finally found the these two command..
.syntax unified
.code 16 
lot of people have these problem and they use these two commands.becasically this issue is regarding to the thumb2 instructions.
but the question is where i need to put these commands
here is refrence of the article 
http://stackoverflow.com/questions/24992876/how-do-i-run-the-linaro-assembler-for-a-thumb2-t32-instruction

juttTechnology, 2016-10-09:
thumb 16 error is go away using the cortax-4 cpu.
here what i got now 

Building target: FruityMesh.elf
Invoking: Cross ARM C++ Linker
arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -Os -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -flto -fno-move-loop-invariants -Wextra  -g -T "C:/nrf/projects/fruitymesh/linker/gcc_nrf52_s132.ld" -Xlinker --gc-sections -L"C:\nrf\workspace/../sdk/ehal_latest/ARM/Nordic/nRF52/CMSIS/Debug" -Wl,-Map,"FruityMesh.map" --specs=nano.specs -L "C:/nrf/projects/fruitymesh/linker" -o "FruityMesh.elf"  ./src/utility/LedWrapper.o ./src/utility/LogTransport.o ./src/utility/Logger.o ./src/utility/NewStorage.o ./src/utility/PacketQueue.o ./src/utility/SimpleBuffer.o ./src/utility/SimplePushStack.o ./src/utility/SimpleQueue.o ./src/utility/Storage.o ./src/utility/Terminal.o ./src/utility/Utility.o  ./src/test/SystemTest.o ./src/test/TestBattery.o ./src/test/Testing.o  ./src/segger_rtt/RTT_Syscalls_GCC.o ./src/segger_rtt/SEGGER_RTT.o  ./src/nrf51/Vectors_nRF51.o ./src/nrf51/app_timer.o ./src/nrf51/ble_radio_notification.o ./src/nrf51/pstorage.o ./src/nrf51/softdevice_handler.o  ./src/nrf/crc16.o ./src/nrf/simple_uart.o  ./src/modules/AdvertisingModule.o ./src/modules/DebugModule.o ./src/modules/EnrollmentModule.o ./src/modules/IoModule.o ./src/modules/Module.o ./src/modules/ScanningModule.o ./src/modules/StatusReporterModule.o  ./src/mesh/Connection.o ./src/mesh/ConnectionManager.o ./src/mesh/Node.o  ./src/base/AdvertisingController.o ./src/base/GAPController.o ./src/base/GATTController.o ./src/base/ScanController.o  ./src/Main.o   -lCMSIS
c:/nrf/sdk/gcc_arm_embedded_4_9/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: error: C:\nrf\workspace/../sdk/ehal_latest/ARM/Nordic/nRF52/CMSIS/Debug\libCMSIS.a(ResetEntry.o) uses VFP register arguments, FruityMesh.elf does not
c:/nrf/sdk/gcc_arm_embedded_4_9/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: failed to merge target specific data of file C:\nrf\workspace/../sdk/ehal_latest/ARM/Nordic/nRF52/CMSIS/Debug\libCMSIS.a(ResetEntry.o)
c:/nrf/sdk/gcc_arm_embedded_4_9/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: error: C:\nrf\workspace/../sdk/ehal_latest/ARM/Nordic/nRF52/CMSIS/Debug\libCMSIS.a(system_nrf52.o) uses VFP register arguments, FruityMesh.elf does not
c:/nrf/sdk/gcc_arm_embedded_4_9/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: failed to merge target specific data of file C:\nrf\workspace/../sdk/ehal_latest/ARM/Nordic/nRF52/CMSIS/Debug\libCMSIS.a(system_nrf52.o)
c:/nrf/sdk/gcc_arm_embedded_4_9/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: warning: cannot find entry symbol Reset_Handler; defaulting to 0001c000
collect2.exe: error: ld returned 1 exit status
makefile:65: recipe for target 'FruityMesh.elf' failed
make: *** [FruityMesh.elf] Error 1

mariusheil, 2016-10-09:
I think I got an idea. The Project includes the ehal project which is precompiled. Guess you have to configure that Project to compile for the nrf52 first. i think floating point unit had to be enabled for both projects for the nrf52

juttTechnology, 2016-10-09:
Sorry for bothering you. your answer makes really sense.
but how i  have to configure ehal Project to compile for the nrf52 first.
ehal project folder was Set up in the Setup file and then just run.
yes in the Setup,  i set the path for the Nrf52.

really no idea.
this Project is really super for Nrf51. but its really hard to setup for Nrf52. i am trying to Run it on Nrf52  from last 1 week but still not successful.
 would be Great if you also have  setup for Nrf52 too.

thank you

mariusheil, 2016-10-12:
Hi, sorry for the nrf52 problems. I know that this is tedious but did not
have time to optimize it since I rarely use the nRF52 right now. You have
to open the CMSIS project from this
path: sdk\ehal_latest\ARM\Nordic\nRF51\CMSIS if it is not already in your
eclipse workspace. Under build settings, I think the ARM family has to be
switched to m4 and the float abi is FP instructions hard and fpv4-sp-d16.
Not sure if changing the Preprocessor defines had to be changed to NRF52,
but you should do that. Afterwards that project should build and should
give you a binary of the CMSIS library which can then be linked with the
FruityMesh project. I think that works automatically from there. If not,
you have to adjust either the include library under the Build Settings of
FruityMesh or adjust the search path in the linker script.

juttTechnology, 2016-10-13:
thank you very Much.
Now i have fruitymesh working on Nrf52 using VisualGDB..i did some changes but finally  successfull.

closed 2016-11-09 by juttTechnology

kkap6, 2017-06-06:
Hello, I am also trying to run FruityMesh on nRF52, @juttTechnology could you write what did you changed?
Thanks in advance.

mariusheil, 2017-06-08:
Are you also trying to use Visual Studio or are you using GCC?

kkap6, 2017-06-08:
I am using GCC and Eclipse IDE.

mariusheil, 2017-06-08:
Are you using the Makefile? Can you post the output of make if you try to compile it?

kkap6, 2017-06-08:
After some changes I successfully compile the project but it doesn't work. While I'm debuging I'm getting: "No source available for "0x1afe0" "
 `ldrb    r0, [r1, #1]`

mariusheil, 2017-06-08:
ok, du you have a terminal connected to the UART pins? So you can see if it boots at least partially? You can also try to experiment with the LED commands to see if at least some code of the Main routine is triggered. Which dev-board are you using?

mariusheil, 2017-06-08:
Did you go into the config.h and have you configured the correct default boardType for NRF52?

kkap6, 2017-06-09:
Yes, I did it. Also I had:
- change processor type to cortex-m4 in Makefile.NRF52
- add to Makefile CFLAGS += -mfloat-abi=hard -mfpu=fpv4-sp-d16

kkap6, 2017-06-14:
Do you have another ideas why it isn't working? Did you maybe try run it on nrf52? Anyway thanks for your help.

skulblakka, 2017-06-14:
Did you take a look at #77? It looks like you need to revert the changes to the nrf5x_common.ld to get a working build for the NRF52.

kkap6, 2017-06-14:
Thanks you @skulblakka, now the main is starting.