bluerange-io/bluerange-mesh #67: Reduce range of BLE ?

anujdutt9, 2016-10-28:
Hi, I am working with your BLE Mesh. 
I would like to change the range of the BLE connection like decrease it so that the nRF dongles connect at a particular distance and disconnect as they go out of that range. I tried changing the  rxTransmissionPower" to -30dB in the Config/Config.h file, the range has decreased but not that much. -40dB dosen't work with the mesh. 
Is there any other factor in the code that I can change to decrease the range of the Mesh ??
Thanks :)

mariusheil, 2016-10-28:
Hi, setting the txPower to -30 is the lowest for the nrf51 and reduces possible connection ranges to about 20cm between two devices.

i am currently working on connection suspension so that a reconnection occurs within a user settable Timeout.

anujdutt9, 2016-10-28:
Yes I noticed that. The connection range is about 20 cm and disconnection range is about 130 cm, depends on the environment. I want to decrease this disconnection range. 
When will you be updating the code with your new features ?
Thanks :)

Informatic, 2016-10-28:
Are you sure transmission power in config is used anywhere in current
master branch? As far as I checked some time ago it wasnt used anywhere for
setting actual transmission power.

anujdutt9, 2016-10-28:
Well according to Nordic it is the only factor to change the transmission power. I have changed it from 0 to -20, -30dB and there is surely a change in connection range. In the config file, they have mapped the value to a parameter "sd_ble_gap_tx_power_set(Config->radioTransmitPower);" which is used at most places in the code. : )

closed 2016-10-28 by anujdutt9