bluerange-io/bluerange-mesh #69: Building issue

xiamaiani, 2016-11-08:
Hello Marius,

I have a question about building FruityMesh.
When i follow all the steps on fruitymesh-devenv en build for the first time i get an error.
Wich is:
```
make: *** No rule to make target 'src/nrf/Vectors_nRF51.o', needed by 'FruityMesh.elf'.  Stop.
make: *** Waiting for unfinished jobs.... 
```
If I remove this file from the project tor problem is solved. 
However it will happen with the next ones aswell:
App_timer
ble_radio_notification
nrf_delay
pstorage
softdevice_handler
vectors_nrf51

All of these files are supposed to be in: C:\nrf\projects\fruitymesh\src\nrf.
But this map has only crc16.c and simple_uart.c in it.

Also when i try to open one of the above eclipse says: File not found: C:\nrf\projects\fruitymesh\src\nrf\pstorage.c.
And the other files say the same, except crc16.c and simple_uart.c.

Should these be removed or do they need to be in the project to run?

Kind regards.

mariusheil, 2016-11-08:
Hi, sorry for the bad documentation. I work with a makefile locally
and I should really upload this to github but am a little short on
time currently :-( The include paths that I use are the following.
Most of the files are in the sdk/components folder:

mariusheil, 2016-11-08:
sorry, sent the e-mail too soon, updated it.

xiamaiani, 2016-11-08:
Hello,

I checked the  makefile but the path I use are the same.
`Finished building: C:/nrf/sdk/nrf51_sdk_latest/components/ble/ble_radio_notification/ble_radio_notification.c
 
make: *** No rule to make target 'src/nrf/Vectors_nRF51.o', needed by 'FruityMesh.elf'.  Stop.
make: *** Waiting for unfinished jobs....`

the building finished, but after that it says no rule to make.

xiamaiani, 2016-11-08:
It seems like he is looking in the wrong directory.
The files he is looking for are in nrf51.
How can i change the directory he should be looking for>

Informatic, 2016-11-08:
https://github.com/mwaylabs/fruitymesh/pull/52 You might be interested in my somehow patched makefile without dependency on ehal.

xiamaiani, 2016-11-09:
Hello Informatic,

When i use your makefiles I had to edit:
`NRF5_SDK_PATH ?= $(HOME)/nrf/sdk/nrf5_sdk_latest`
to match my folder wich was
 `NRF5_SDK_PATH ?= $(HOME)/nrf/sdk/nrf51_sdk_latest.`
After that i try to build but get the error:
`Making _build
mkdir  > nul 2>&1 || (exit 0) _build
c:/nrf/sdk/gnu_arm_eclipse_build_tools_2_4/bin/sh: syntax error: unexpected word
make: *** [_build] Error 2
Makefile:175: recipe for target '_build' failed`

Any ideas?

Thanks in advance

Informatic, 2016-11-09:
First - current master branch requires "nRF5 SDK" version 11, afair.
Second - I'll look into it later today, seems like windows support is
somehow broken. (I'm not really a Windows user myself) I'm pretty sure
Marius reported that it's working fine on Windows.

xiamaiani, 2016-11-09:
I downloaded version 11 from nRF5 SDK
https://www.nordicsemi.com/eng/nordic/Products/nRF5-SDK/nRF5-SDK-zip/54291
I unpacked it in C:\nrf\sdk\nrf_sdk_11_0
After this i edit the setup.bat to point at the new map since it says nrf_sdk_9_0 in it.
I change the location aswell to point at the right folders.
After I run setup.bat it names the new map nrf51_sdk_latest.
Am I doing something wrong?

closed 2016-11-10 by xiamaiani