bluerange-io/bluerange-mesh #71: S130 V2 16kb Flash and Ram setting

xiamaiani, 2016-11-15:
Hello,

I cant seem to figure out that the right setting are for the RAM and FLash.

I currently use 
 `MEMORY
{
  FLASH (rx) : ORIGIN = 0x1B000, LENGTH = 0x24000
  RAM (rwx) :  ORIGIN = 0x20002400, LENGTH = 0x1C00
}`

But then the he writes:
`[Main.cpp@236 NODE]: Softdevice Init OK                                    
[Main.cpp@261 ERROR]: Ram base at 0x20002400                  
[Main.cpp@270 ERROR]: Fatal: Not enough memory for the selected configuration. Required:0x20002e80     
`
What configuration should i use?

mariusheil, 2016-11-15:
Can you try this address? 0x20002e80

xiamaiani, 2016-11-15:
Hello,

When i use these setting, nothing happens. It seems like the program doesnt run.
  FLASH (rx) : ORIGIN = 0x1b000, LENGTH = 0x24000
  RAM (rwx) :  ORIGIN = 0x20002E80, LENGTH = 0x1C00
I dont get a response in the terminal.

xiamaiani, 2016-11-16:
Hello,

I tried to use 
  FLASH (rx) : ORIGIN = 0x1B000, LENGTH = 0x25000
  RAM (rwx) :  ORIGIN = 0x20002e80, LENGTH = 0x1180
As  you said,
However i get hte error:
`c:/nrf/sdk/gcc_arm_embedded_4_9/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: FruityMesh.elf section `.bss' will not fit in region `RAM'
c:/nrf/sdk/gcc_arm_embedded_4_9/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: region RAM overflowed with stack
c:/nrf/sdk/gcc_arm_embedded_4_9/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: region `RAM' overflowed by 1544 bytes`

Does this mean that fruitymesh cant be runned on the 16kb version?

closed 2017-01-10 by xiamaiani