bluerange-io/bluerange-mesh #72: Mesh_Out_Connection from 3 to 7

juttTechnology, 2016-11-21:
Hello,
         i have one question.i am using the Nrf52.
         when i try to change the MESH_OUT_CONNECTIONS 7 then i got problem.
         first i fix the  __application_ram_start_address. 
         
         Now i am using the 
         u32 __application_ram_start_address = 0x20003F30;

         Now ram start Address have no error. but he is crashing inside of the 
                           GATTController::bleMeshServiceInit
         on the line
                     err = sd_ble_uuid_vs_add(&baseUUID128, &meshService.serviceUuid.type);
       i got the err number 16. which is bad memory address.

any suggestions would be nice.
thank you

mariusheil, 2016-12-13:
Hello,

did you succeed? Take a look at the sd_ functin that uses the number of connections. I think they documented, which parameters you need to update to assign more memory to the softdevice. Maybe you have to give some more space when initializing the attribute table, but I am not sure.
6LowPan is a different layer than BLE. Fruitymesh works on top of BLE. On top of 6LowPan you have to have an implementation that defines meshing such as ZigBee or Thread, but both these protocols assume that you have power-connected devices that do the routing and meshing. The BLE mesh standard seems to follow down the same route as I've heard. FruityMesh doesn't assume that you have power connected devices and allows all devices to route traffic, but it is possible to distribute the power consumption to devices with more energy when choosing so.

freelinuxer-cn, 2016-12-29:
nrf52 ： #define MESH_OUT_CONNECTIONS 7 is fine，change the ld script. 
MEMORY
{
  FLASH (rx) : ORIGIN = 0x1c000, LENGTH = 0x60000
  RAM (rwx) :  ORIGIN = 0x20004300, LENGTH = 0xbd00

  BOOTLOADER (rx) : ORIGIN = 0x7c000, LENGTH = 0x4000
}

closed 2016-12-31 by juttTechnology