bluerange-io/bluerange-mesh #73: Data rates improvement question

xiamaiani, 2016-11-24:
Hello,

I was doing some test with the data rate from FruityMesh.
What i noticed was that the data rates were slower then expected.

I changed the bandwidth to high;
`    conn_bw.conn_bw.conn_bw_rx = BLE_CONN_BW_HIGH;
    conn_bw.conn_bw.conn_bw_tx = BLE_CONN_BW_HIGH;`

i changed the interval to 7,5 ms
`		u16 meshMinConnectionInterval = MSEC_TO_UNITS(7.5, UNIT_1_25_MS);  
		u16 meshMaxConnectionInterval = MSEC_TO_UNITS(7.5, UNIT_1_25_MS); `

After this send 20 bytes of data, 25 times after the command "datal" with relaible on false.
`		for(int z = 0; z < 25; z++)
		{
			cm->SendMessageToReceiver(NULL, &data[0], 20, false);
		}`
The time it took to get from a nrf52 board one to board 2 was 2,7 seconds.
500 bytes in 2.7 seconds == ~180bytes/s 
but when you calculate the speed it should reach around 2660 bytes/s
This is when you calculate only 1 packet per event instead of 6.

Any ideas how to improve this data rate speed?

Thanks in advance.

mariusheil, 2016-11-24:
Have you made sure that the data was really sent unreliably? E.g. if you have packets bigger thatn 20 bytes (maybe because of the additional packet header?) They will get sent reliably because the implementation forces this.
I have made tests with the settings in the repository and depending on the role (peripheral or central), there were 2 or 6 transmit buffers free. You can check the transmit buffers available with the status command.
Using a sniffer, I could observe that all transmit buffers were used and multiple packets were sent per connection interval.
if you type "debug CONN_DATA", you'll see how the data was received.

xiamaiani, 2016-11-24:
The terminal gave this back for every packet:
Received type 42, hasMore 0, length 20, reliable 2
hasMore is 0, but why is reliable 2?
also the packet queue was empty. for both devices before sending the data.

mariusheil, 2016-11-24:
That stands for BLE_GATTS_OP_WRITE_CMD which is an unacknowledged packet.
So that looks fine, you are sending unreliable packets. I think you have to
use a packet sniffer to observe the traffic more closely to find the speed
issue. Best to only enable advertising channel 37 in the config, then it
will be easier for the nordic sniffer to pick up the conneciton event
between two nodes. (just reconnect them a few times until the sniffer picks
up the data flow.)
What does the status command say on both nodes? the txBuffers are the
interesting part it should tell you the number of reliable and unreliable
buffers granted by the softdevice. Maybe 7.5ms interval is too high for
multiple packets per event if you want to have connections between multiple
nodes. But I would have to read through the nordic docs to find out.

You could try a slower connection interval like 100ms and do that
calculation again.

xiamaiani, 2016-11-24:
I restarted both devices,
First check on both devices without sending anything was:
board 1 =  OUT 4536, state:4, clId:bcfb2fe2, clSize:1, toSink:-1, Queue:33-33(0), Buf(rel
:1, unrel:1), mb:1, pend:0
board 2=  < IN  12258, state:4, clId:bcfb2fe2, clSize:1, toSink:-1, Queue:10-10(0), Buf(re
l:1, unrel:7), mb:0, pend:0>

After I send the 25 packets unreliable, from board 1 to 2;
board 1 =  OUT 4536, state:4, clId:bcfb2fe2, clSize:1, toSink:-1, Queue:583-583(0), Buf(r
el:1, unrel:1), mb:1, pend:0
board 2 = < IN  12258, state:4, clId:bcfb2fe2, clSize:1, toSink:-1, Queue:10-10(0), Buf(re
l:1, unrel:7), mb:0, pend:0

When i set the max and the min interval to 100 the data rate slowed down.
when i set  the max to 100 and min to 7.5 the time went back to 2.7 seconds

xiamaiani, 2016-11-24:
Hello,

As for the BLE sniffer,
I can see the advertisements messages but what should i be doing with them?
I do not see my messages i send.
Any suggestions to fix this?

xiamaiani, 2016-11-25:
Where can I find your test result, and the way you tested FruityMesh?

xiamaiani, 2016-11-28:
Hello,

I tested some more with the data rate.
I sent 25 times 20 bytes. But between the first and the second message there is about a second delay.
```
for(int z = 0; z < 25; z++)
{
	cm->SendMessageToReceiver(NULL, data, 20, true);
}
rtc first received message: 0x60842
rtc second received message: 0x67EFC
difference = 0x76BA = 30396/32768 = 0,928 seconds.

Any idea why this is happening?

```

xiamaiani, 2016-11-30:
Hello,

I found out why it was so slow.
I was logging the data in a terminal. this will cost extremely much time.
I logged a lot less, now the data rates have increase by 10 times.

closed 2016-11-30 by xiamaiani

mariusheil, 2016-11-30:
Perfect, good finding 😊