bluerange-io/bluerange-mesh #74: Question about ChangeState

xiamaiani, 2016-12-13:
Hello,

I was wondering what the point is of this function?, since it always will remain high.
if (newState == discoveryState::DISCOVERY)
	{
		nextDiscoveryState = discoveryState::INVALID_STATE;

		//Use Low instead of High discovery if no nodes have been found for a while
		if (noNodesFoundCounter < Config->discoveryHighToLowTransitionDuration)
		{
			ChangeState(discoveryState::DISCOVERY_HIGH);
		}
		else
		{
			ChangeState(discoveryState::DISCOVERY_LOW);
		}

		ChangeState(discoveryState::DISCOVERY_HIGH);
	}

mariusheil, 2016-12-13:
Hello,

you are right, that doesn't make sense, I should have left a comment. This was for debugging purpose so that discovery always remeins in the HIGH state,

closed 2016-12-13 by mariusheil

mariusheil, 2016-12-14:
Hi,

I do not think that is the reason, however it might be possible. I am considering going away from the discovery states anyway because it is a little awkward with the seperation with modules (modules should only change the advertising messages while in the BACK_OFF) state. But therefore, I need to implement switching between multiple advertising messages. Sadly this is not easily possible with the Nordic stack.

I noticed some similar results with the timing but I did not do much research in that direction. I would be very interested if you can find the cause of it. There is also some weird disconnection going on between nodes that I think is caused by the softdevice, but it could be an error on my part as well.

xiamaiani, 2016-12-14:
I did some more tests.
This is my setup:
I use 2 NRF52 devices 1 for sending 1 for receiving.
I am sending 100 packages with 20 bytes of payload.
7,5 ms connection interval and high bandwidth. Since it only is going to send and receive it will send 7 packages per connection interval. i used the default code.
I repeated this test 100 times. 
[image]

I did this wole test again but now with DISCOVERY OFF
[image]

Seems like the discovery is making the sending of the data slower.
Another question, When the state changes does that mean that the program instantly start looking for advertisement  messages? or does the interval need to wait a cycle before kicking in?

mariusheil, 2016-12-14:
Mhh, ok. I guess it's like that: depending on the settings of the softdevice advertising and softdevice scanning (window, interval, etc,...) the Softdevice will have to schedule its timings so that it cannot cope with your high connection interval and will have to drop some of the connection events. You could try to lower the scan window size and it should make a difference.

A state change will immediately trigger the Softdevice's advertising function as far as I can remember, you should be able to track it down in the code. However, the time when the softdevice really starts looking for messages is up to the softdevice and there's no access to the internal schedule sadly.

xiamaiani, 2016-12-15:
>  There is also some weird disconnection going on between nodes that I think is caused by the softdevice, but it could be an error on my part as well.

Do the nodes reconnect after this disconnect?

mariusheil, 2016-12-15:
Yes, they do. I have partly implemented some more advanced reconnection which I haven't pushed yet, but the disconnection shouldn't happen in the first place, e.g.:
There is a GATT timeout event that will basically render the GATT layer useless, so the nodes need to reconnect. This happens when a message is not acknowledged after 30 seconds. But the weird thing is that the connection timeout is set to 6 seconds, meaning that it should really disconnect a lot earlier. I haven't yet found the couse for these issues and I hope they are not outside my control in the Softdevice.

xiamaiani, 2016-12-16:
I did some more tests.
i removed the swapping between low and high interval and used a constant value.
```
u16 meshScanIntervalHigh = MSEC_TO_UNITS(200, UNIT_0_625_MS);	
u16 meshScanWindowHigh = MSEC_TO_UNITS(10, UNIT_0_625_MS);	
```
Now the time it takes to send 100 packets of 20 bytes is lower then 200ms.
also the massive differences are gone now which is great.
However the time it takes to connect two device also increased quite alot.

It seems to take quite some testing to get the optimal settings.
Thanks for the support you gave its great :)

mariusheil, 2016-12-16:
Hi,

a decision to connect to another node is only made when switching to the decision state! if you Are staying in the discovery state, This state is never triggered. I will remove this logic when I have time.

xiamaiani, 2016-12-17:
Hello,

can you explain what the exact difference is between
```
u16 meshAdvertisingIntervalHigh = MSEC_TO_UNITS(100, UNIT_0_625_MS);
u16 meshScanIntervalHigh = MSEC_TO_UNITS(20, UNIT_0_625_MS);	
u16 meshScanWindowHigh = MSEC_TO_UNITS(4, UNIT_0_625_MS);	
```
I feel like the first is for advertisement messages.
The second is for mesh messages.
And the third is for how long it looks for those messages for both?

If so, what is the point in this, if there are new times set to look for messages?
```
u16 meshMinConnectionInterval = MSEC_TO_UNITS(7.5, UNIT_1_25_MS);   	
u16 meshMaxConnectionInterval = MSEC_TO_UNITS(7.5, UNIT_1_25_MS);   
```