bluerange-io/bluerange-mesh #75: Uart only getting first 5 bytes from terminal

xiamaiani, 2017-01-10:
Hello, 

I am doing some test with the UART and noticed when i try to send more then 5 bytes/characaters its waiting for a second message it seems?
But when i send 5 or smaller its works fine.
Any idea how to remove this, so it works with bigger messages.

mariusheil, 2017-01-10:
Can you give me a code sample? I am not exaxtly sure what you mean. There are two uart modes, interrupt based and blocking, depending on the terminal mode you use. I guess the Hardware buffer of the uart peripheral was around 5 bytes, but I am not sure what you are trying to do.

xiamaiani, 2017-01-10:
When i use FruityMesh the way it is, and i send 5 characters to FruityMesh it will be handled immediately. however when i send 6 characters, it only registers the first 5 and waits until i send my next  characters. however the second time i send my data it can be larger then 5 bytes.
So i can send 6 bytes after that and it will wait, i send 10 after that and it will see 16 bytes. but i cant seem to remove this waiting around intul the next data arraives.

mariusheil, 2017-01-11:
Mhh, so you use the terminal. But the terminal will always wait until it receives a linebreak (\r) only then will it process the line. Can you try to use putty as well for testing? Are you trying with stopterm or with startterm? (Is the data echoed back to you to the terminal or not? Does it start with mhTerm: ?)

xiamaiani, 2017-01-11:
When i use putty it works fine but thats only because i send one byte at a time.
With realtearm i can send multiple bytes in a row. 
[image]
Thats when i can only send 5 bytes to get an isntant reply. if i send 6 bytes i need to write more to get a response, as if its waiting for a second part.

mariusheil, 2017-01-11:
Hi, that's probably because of the 6 byte receive buffer of the UART Peripheral in the nRF51. Have a look at 29.5 in the nrf51 series reference manual. Are you using stopterm or startterm? Because these use two different implementations. One is interrupt based, the other one is not. Please have a look in Terminal.cpp and try to enter startterm or stopterm in the terminal and see if the behaviour changes. When Using stopterm, you will not get your characters echoed back!

xiamaiani, 2017-01-11:
Hello, thanks for the response,
It seems that indeed the buffer is the issue. i changed \r to 0x61 as trigger
When i send:
0x50 0x51 0x52 0x53 0x54 0x55 0x56 0x57 0x61 nothing happens.
then i send
0x58 0x59 0x60 0x63 0x64 0x65 0x66 0x67 0x61
Then i get this back from the nrf52.
50 51 52 53 54 55 58 59 60 63 64 65 66 67
only the first six characters are read by the device, however the second message has been read whole. i assume this is because the program is in a while loop until he enters this if statement
if (byteBuffer == 0x61 || counter >= readBufferLength || counter >= 250)

xiamaiani, 2017-01-11:
The weird thing is, I read that: The UART receiver chain implements a FIFO capable of storing six incoming RXD bytes before data is overwritten.
That means that the of the first message this part would be read: 0x53 0x54 0x55 0x56 0x57 0x61, since it would overwrite the character in that buffer. that is not the case in this instance, the last part is being ingnored.

mariusheil, 2017-01-11:
Hi,

I just saw that I have a lot of code here that I couldn't yet push to github. And the Terminal is one of these. I have uploaded the files including my current Main.cpp where you see how the Terminal should be integrated. I hopfe that this is easy. I'll look through my current code and see what I have to do to push it to github. Sorry for that.

[terminal.zip](https://github.com/mwaylabs/fruitymesh/files/698369/terminal.zip)

xiamaiani, 2017-01-11:
thanks, i will have a look into it.

xiamaiani, 2017-01-11:
Hello,

Thanks for the new files, It works a lot better now, most of the time i receive all bytes. Sometimes it still has the issue that it only receives 6 bytes, i work on a baudrate of 921600, that might be also a problem. maybe a setting somewhere?

mariusheil, 2017-01-11:
did it! you can now simply take the new release

xiamaiani, 2017-01-11:
Hello,

I  implemented you new code!
It works great now. 

Thanks!

closed 2017-01-11 by xiamaiani