bluerange-io/bluerange-mesh #76: UART doenst work when i use a battery instead of microusb

xiamaiani, 2017-01-24:
Hello,

I am using UART, but when I use a battery instead of the microusb my aurt doenst work anymore.
It only works when i connect the nrf52 dk to a computer. I am using pin 6 and 8 with a rs232 cable. I am using the right COM port to read the data. Any idea what the issue might be? Some init only when SEGGER is active?

Informatic, 2017-01-24:
Keep in mind UART by default uses hardware flow control as well as
normal RX/TX pins, so in the end you need 4 pins.

You should be able to disable hardware flow control (HWFC) if you really
need to here:
https://github.com/mwaylabs/fruitymesh/blob/master/src/utility/Terminal.cpp#L262

I haven't really tested it myself. I think SoftDevice may cause some
tiny issues when transmitting/receiving data on UART without HWFC.

xiamaiani, 2017-01-24:
Thank you very much,
You are a lifesaver. It works great now.

closed 2017-01-24 by xiamaiani