bluerange-io/bluerange-mesh #79: Device restart randomly with TerminalCallbacks

Bardo91, 2017-06-07:
Hi, first of all thanks for all your work. It is a cool library for using nordic devices!

I am trying to send messages from a computer to another through  nRF51Dongles. It works in general but when I send various messages through the serial port  to be handled in the modules and sent through the network it seems that the device gets restarted. Then once it is connected again to the mesh it start sending messages again but after 2 or 3 messages it gets restarted again (and so on...)

I was trying to find where the error happens and the code is executed properly until it reaches:
```
	for(u32 i=0; i<registeredCallbacks->size(); i++){
		handled += ((TerminalCommandListener*)registeredCallbacks->GetItemAt(i))->TerminalCommandHandler(commandName, commandArgs);
	}
```

This is my callback:

```
//---------------------------------------------------------------------------------------------------------------------
bool CustomModule::TerminalCommandHandler(string commandName, vector<string> commandArgs) {
	if(commandName == "custommod") {
		if(commandArgs[0] == "send_lat" || commandArgs[0] == "send_lon"){
			std::string val = "NAN";
			if(commandArgs.size() == 2){
				val = commandArgs[1];	// latitude
			}else{
				logt("CUSTOMMOD", "Error, need extra argument with data to sent");
				return false;
			}
			
			connPacketModule* outPacket = new connPacketModule();
			outPacket->header.messageType = MESSAGE_TYPE_DATA_1;
			outPacket->header.sender = node->persistentConfig.nodeId;
			outPacket->moduleId = moduleId;
			outPacket->actionType = CustomModuleTriggerActionMessages::TRIGGER_MESSAGE;
			
			std::string data;
			if(commandArgs[0] == "send_lat")
				data = "///L"+val+"///";	
			else if(commandArgs[0] == "send_lon")
				data = "///N"+val+"///";	

			memcpy(&outPacket->data, data.c_str(), data.size()*sizeof(char));
			int totalSize = SIZEOF_CONN_PACKET_MODULE + sizeof(char)*data.size();

			for(int con = 0; con < 4; con++){	// Cover 4 possible connections
				if(cm){
					if(cm->connections[con]){
						if(cm->connections[con]->isConnected()){
							outPacket->header.receiver = cm->connections[con]->partnerId;
							cm->SendMessageToReceiver(NULL, (u8*) outPacket, totalSize, true);
						}
					}
				}
			}
			delete outPacket;
			return true;
		}
	}
	//Must be called to allow the module to get and set the config
	return Module::TerminalCommandHandler(commandName, commandArgs);
}
```

As I said, it works in general, but if I send various messages sequentially the dongle restarts and the communication stops (and it takes some seconds to reconnect) .

I think that might be related with calling the callback several times before the previous ends, but I am not sure if it is multithreading or not and I cant debug it properly.

Do anyone have any clue of what is happening? Thanks in advance

mariusheil, 2017-06-08:
Hi,

did everything work before you had your custom module integrated? Can you execute other commands that send data between nodes? Debugging is usually complicated but what you can do is go in the config and enable the DEBUG switch. Then you should be able to put breakpoints in the fault handling routines that you can find in the Main.cpp (e.g. HardFault Handler, Assert Handler,..) As soon as the node crashes, it will get stuck in an endless loop and your breakpoint should trigger together with a stacktrace. You can also enable logging if not yet done and log sth. every few lines to find the error.
Pay attention that the data_1 packet only has 14 bytes of possible payload as the MTU is 20. When your string gets longer, your memcpy will overwrite parts of your stack. You should also do a quick check what the receiving function does with the DATA_1 packet. Maybe it always expects it to be fixed length with 20 bytes.

Marius

mariusheil, 2017-06-10:
> Hi Marius, that fixes the problem! As you said, it was related with the size of the packet. I was exceeding the max. 14 bytes. I was encoding a couple of floats in an array of chars. Now I am memcpying the floats in the array of bytes and it doesn't restart as they are always 4+4 bytes.
> Thank you so much.
> Before closing the issue; I am struggling with a minor issue now. I am trying to send the data through the terminal using "logt" function. I tried doing something like: logt("CUSTOMMOD", "value: %f", value); but it doesn't work with the %f, it puts an empty char in its place. If I cast the value to int and put "%d" it works (but I lose the decimals).
> I tried using "sprintf" and "snprintf" and then use "%s" in the logt(...) but the same thing happens. Is it any reason why the printf family of function doesn't work with "%f" or am I messing up with something?.

Hi,

the nrf51 does not have a Floating point unit, so that code is added by the compiler, you should be aware of this and possibly only use integer arithmetic if this is possible. I do not know if that is the Problem in your case, though.

Bardo91, 2017-06-10:
Definitely, that is probably the problem. I will figure out how to change what I am doing using only integer arithmetic. 

Thank you very much again, I am closing the issue.

closed 2017-06-10 by Bardo91