bluerange-io/bluerange-mesh #84: nRF52.h's NRF_UICR_Type doesn't have BOOTLOADERADDR

onionhammer, 2017-06-18:
Causes compile issue here
 https://github.com/mwaylabs/fruitymesh/blob/master/src/modules/Module.cpp#L68

skulblakka, 2017-06-18:
I think they removed the ```BOOTLOADERADDR``` in the UICR struct and the address is now set in
```NRF_UICR->NRFFW[0]```. 

However I'm not completely sure as I could not find the correct information in the SDK 11 documentation (according to [the SDK 12 documentation](https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v12.0.0/bootloader_memory_nrf52.svg) ```NRF_UICR->NRFFW[0]``` should equal the bootloader start address). Also I'm not making use of a bootloader. While this change works without one, I don't know for sure if it works with one.

mariusheil, 2017-06-19:
yes, you are right. NRF_UICR->NRFFW[0] is the correct one for the nrf52

closed 2017-08-09 by mariusheil