bluerange-io/bluerange-mesh #86: error: stray '\' in program SVCALL

asoftplus, 2017-08-14:
D:/nrf/sdk/nrf_sdk_11_0/components/softdevice/s130/headers/ble_gap.h:988:1: error: stray '\' in programSVCALL(SD_BLE_GAP_ADDRESS_SET, uint32_t, sd_ble_gap_address_set(uint8_t addr_cycle_mode, ble_gap_addr_t const *p_addr));^D:/nrf/sdk/nrf_sdk_11_0/components/softdevice/s130/headers/ble_gap.h:998:1: error: stray '\' in programSVCALL(SD_BLE_GAP_ADDRESS_GET, uint32_t, sd_ble_gap_address_get(ble_gap_addr_t *p_addr)); ^D:/nrf/sdk/nrf_sdk_11_0/components/softdevice/s130/headers/ble_gap.h:1031:1: error: stray '\' in program SVCALL(SD_BLE_GAP_ADV_DATA_SET, uint32_t, sd_ble_gap_adv_data_set(uint8_t const *p_data, uint8_t dlen, uint8_t const *p_sr_data, uint8_t srdlen)); ^D:/nrf/sdk/nrf_sdk_11_0/components/softdevice/s130/headers/ble_gap.h:1070:1: error: stray '\' in program SVCALL(SD_BLE_GAP_ADV_START, uint32_t, sd_ble_gap_adv_start(ble_gap_adv_params_t const *p_adv_params)); ^D:/nrf/sdk/nrf_sdk_11_0/components/softdevice/s130/headers/ble_gap.h:1083:1: error: stray '\' in program-- I am trying to build FruityMesh. I have got of a lot of "stray '\' in program" errors. Do you know what it means? My colleague said that it may be due to a macro problem. So I go to #define SVCALL(number, return_type, signature)          \  _Pragma("GCC diagnostic push")                        \  _Pragma("GCC diagnostic ignored \"-Wreturn-type\"")   \  __attribute__((naked))                                \  __attribute__((unused))                               \  static return_type signature                          \  {                                                     \    __asm(                                              \        "svc %0\n"                                      \        "bx r14" : : "I" (GCC_CAST_CPP number) : "r0"   \    );                                                  \  }                                                     \  _Pragma("GCC diagnostic pop")  But I still have no clue on how to solve this problem. Thanks for your attention.

mariusheil, 2017-08-14:
Take a look at settings up the development environment, there's a bug in the nRF5 SDK in v11: https://github.com/mwaylabs/fruitymesh-devenv#fix-some-of-the-sdk-files

asoftplus, 2017-08-14:
Thanks for your quick reply.

I have tried those steps, but still gets the same errors.

Any other possible ways to solve the problem?

mariusheil, 2017-08-14:
Are you using the recommended versions of the sdk and the GCC compiler? Can you have a look at the preprocessor output to see what's being generated? Did you clean the project before rebuilding it?

asoftplus, 2017-08-15:
Thanks for your help. I have finally solved the problem and successfully complied FruityMesh.

The documentation may have a minor bug. 

In the step 3, it writes

> You have to ... at line 55...
>> #define GCC_CAST_CPP (uint16_t)  \

After removing the ending backslash, the bug is solved.

mariusheil, 2017-08-15:
You are right, just checked it! Sorry for that, I changed the documentation and removed the backslash.

closed 2017-08-15 by mariusheil