bluerange-io/bluerange-mesh #87: Sees nothing in the Terminal

asoftplus, 2017-08-15:
After downloading the softdevice and FruityMesh to my chip, I can see nothing in the Terminal. 

Is there anything I can do to locate the source of the problem?

skulblakka, 2017-08-16:
Did you take a look at #77? (in short you might have to revert to an earlier version of nrf5x_common.ld)

asoftplus, 2017-08-18:
Thanks a lot for your hint. I have not read issue 77 in detail yet. But I will do it today when I go to work. 

However, I did try to revert nrf5x_common.ld to an earlier version. 

The difference is this: 

The original nothingness is: 

[][][] 

When I try to commands such as "status", only another new line of [][][] appears. 

After reverting nrf5x_common.ld to an earlier version, the console screen is totally blank, not even [][][]. 

Does this change of behavior mean some clues for solving this issue?

skulblakka, 2017-08-18:
The issue caused by #77 is that the current version of nrf5x_common.ld in combination with a nrf52 DK prevents the board from correctly booting up, meaning that you will see no output at all with the current version.

The fact that you get an output with the current version makes me think the issue might be elsewhere.

btw what chip are you using?

asoftplus, 2017-08-19:
Thanks with your further hint. 

I think that you are correct, since I am using nRF51822 instead. 

My colleague have found me this description in the documentation: 

> The memory requirements of the current implementation are not yet optimized much, which is why there is some effort necessary to run it on 16KB RAM chips. If some of the buffers are reduced and the terminal and logging are disabled in the config, FruityMesh will run with 16kb ram devices and the binary will be around 30kb.

So I suspect that my issue is the same as issue #71. However, that issue does not contain the final solution at the end. So my colleague has ordered a 32KB version of nRF51822 chip. 

It will arrive next week. I will test it then. 

Thanks again for your time.

asoftplus, 2017-08-25:
My 32KB board has arrived. It works. I can see terminal output from FruityMesh. 

Additional note: We should use Putty, not other terminal software, because I cannot see the full result from other terminal software.

closed 2017-08-25 by asoftplus