bluerange-io/bluerange-mesh #88: Link Error - invalid assignment to location counter

BeaconZone, 2017-08-21:
I have been trying to build FruityMesh on Windows. The Master branch gives the following link error:

 'C:/Program Files (x86)/GNU Tools ARM Embedded/6 2017-q2-update/bin/arm-none-eabi-g++' -mcpu=cortex-m0 -mthumb -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -flto -fno-move-loop-invariants -Wextra -DBLE_STACK_SUPPORT_REQD -D DEBUG -Os -g -DNRF51_BOARD -DNRF51 -D__need___va_list -Llinker/ -Tlinker/gcc_nrf51_s130_32kb.ld -Xlinker --gc-sections -Wl,-Map,"_build/debug/NRF51_BOARD/FruityMesh.map" --specs=nano.specs _build/debug/NRF51_BOARD/ble_radio_notification.o _build/debug/NRF51_BOARD/nrf_drv_common.o _build/debug/NRF51_BOARD/pstorage.o _build/debug/NRF51_BOARD/nrf_drv_gpiote.o _build/debug/NRF51_BOARD/app_timer.o _build/debug/NRF51_BOARD/softdevice_handler.o _build/debug/NRF51_BOARD/simple_uart.o _build/debug/NRF51_BOARD/crc16.o _build/debug/NRF51_BOARD/RTT_Syscalls_GCC.o _build/debug/NRF51_BOARD/SEGGER_RTT.o _build/debug/NRF51_BOARD/system_nrf51.o _build/debug/NRF51_BOARD/Config.o _build/debug/NRF51_BOARD/Main.o _build/debug/NRF51_BOARD/ScanController.o _build/debug/NRF51_BOARD/GAPController.o _build/debug/NRF51_BOARD/GATTController.o _build/debug/NRF51_BOARD/AdvertisingController.o _build/debug/NRF51_BOARD/Node.o _build/debug/NRF51_BOARD/ConnectionManager.o _build/debug/NRF51_BOARD/Connection.o _build/debug/NRF51_BOARD/ScanningModule.o _build/debug/NRF51_BOARD/IoModule.o _build/debug/NRF51_BOARD/StatusReporterModule.o _build/debug/NRF51_BOARD/Module.o _build/debug/NRF51_BOARD/EnrollmentModule.o _build/debug/NRF51_BOARD/AdvertisingModule.o _build/debug/NRF51_BOARD/DebugModule.o _build/debug/NRF51_BOARD/Testing.o _build/debug/NRF51_BOARD/TestBattery.o _build/debug/NRF51_BOARD/SystemTest.o _build/debug/NRF51_BOARD/SimpleBuffer.o _build/debug/NRF51_BOARD/SimplePushStack.o _build/debug/NRF51_BOARD/Utility.o _build/debug/NRF51_BOARD/Terminal.o _build/debug/NRF51_BOARD/Logger.o _build/debug/NRF51_BOARD/LedWrapper.o _build/debug/NRF51_BOARD/ButtonListener.o _build/debug/NRF51_BOARD/NewStorage.o _build/debug/NRF51_BOARD/Storage.o _build/debug/NRF51_BOARD/SimpleQueue.o _build/debug/NRF51_BOARD/PacketQueue.o _build/debug/NRF51_BOARD/gcc_startup_nrf51.o -lgcc -lc -lnosys -o _build/debug/NRF51_BOARD/FruityMesh.out
nrf5x_common.ld:130 invalid assignment to location counter
collect2.exe: error: ld returned 1 exit status
Makefile:207: recipe for target '_build/debug/NRF51_BOARD/FruityMesh.out' failed
make: *** [_build/debug/NRF51_BOARD/FruityMesh.out] Error 1

Anyone know how to fix the "invalid assignment to location counter" error?

Thanks

Simon

mariusheil, 2017-08-23:
Interesting, how many different errors one could get from a single branch,...
Just to make sure: You used GCC 4.9 and the nRF SDK 11?

BeaconZone, 2017-08-23:
I wasn't using GCC 4.9. It now builds ok thanks.

closed 2017-08-23 by BeaconZone