bluerange-io/bluerange-mesh #90: Uart often stops

asoftplus, 2017-09-12:
Dear all, 

My application needs to keep outputting uart signals indefinitely. However, I have found that the uart signal would stop after a random time. Sometimes 15 minutes, sometimes 30 minutes. 

I had to turn on PuTTY to input something to trigger the uart to emit signals again. Since for my application manual operation is not allowed, I had to set the chip to reboot itself every 15 minutes. 

This workaround sometimes works, sometimes not. 

Would you mind suggesting me on how to solve this bug? 

(I have already turn off the hardware flow control.)

Thanks a lot.

mariusheil, 2017-09-12:
Have a look at the Terminal.cpp file. There is a function
called UartPutStringBlockingWithTimeout which turns off the terminal if a
timeout occured during a write. I've recently fixed this bug in an internal
build. Try to comment out the line that stops the terminal. It was a rather
bad idea ;-)

asoftplus, 2017-09-12:
Thanks for your advice. 

I will try it tomorrow.

closed 2017-09-12 by asoftplus