bluerange-io/bluerange-mesh #92: Power consumption measurements for energy harvesting application

Dving16, 2017-10-25:
Dear all

We are working on a project where we want to look at the possibility to utilise ble mesh networks with energy harvesting. 

We have a working fruity mesh up and running and are now working on a factorial design to determine optimal parameters with discovery on to get the lowest possible power consumption for our application. 
we are following the advise given in #91 to achieve optimal power levels.

Next we want to look at power consumption when discovery is off. 

What would be the best way to measure power drain when sending data through the network?

For this we think of two different approaches. 
- Using a multimeter giving us a average system power measurement, for a specific data package sendt with a specific interval.
- oscilloscope for current profile measurement to determine power needs when sende data packages.

What would be the best way to send data packages emulation sensordata going through the network?

Best Regards

Rasmus
AAU Copenhagen

mariusheil, 2017-10-26:
Have a look at the status reporter module, you can configure it to send periodic status messages, these fit within 1 packet. You can specify the interval at which they are sent.

Dving16, 2017-10-26:
Thank you for your quick reply. I looks like a soothed solution.

Just a question in terms of trying to optimise the power handling in the system for my specific application. Could transmitpower in the config.h file be something I could optimist as I have a static setup with the nodes? By default in the code  radioTransmitPower = 4 which must be equevelent to the board maximum of +4dBm. 

Reducing this value could increased battery capacity and reduced interference problems as I understand it.

mariusheil, 2017-10-26:
Yes, that will help a lot. You can change it at runtime if you wish as well, the advertising module does sth. like that, it calls sd_ble_tx_power_set.

Dving16, 2017-10-30:
Is it possible to enable and disable DC/DC and UART during runtime?

Dving16, 2017-11-08:
Could you point me in the direction on how to change the radioTransmitPower at runtime?

In the beginning of the description og battery consumption measurements in the wiki initial timing is tested for "sleep doing nothing" the final value is set to 500 m/s

Where is the timer interval values change in the script? 

-Regards

Rasmus

mariusheil, 2017-11-14:
Hi, I am not sure if I understand. You have to call the function sd_ble_tx_power_set to change the tx power. Do this anywhere you like and look through the code where this function is used.

The TestBattery does not contain all tests described in the wiki, the timer value was sometimes changed by hand and then recompiled. You will find the timer value in the config.h but it doesn't make much of a difference how often the timer is executed in terms of battery usage. This is why it is currently set to an interval which is good for most use-cases.

Dving16, 2017-11-15:
Thanks for your answer. 

We are slowly getting better knowledge of the functions as we go. But it is a slow pace as we are mechanical engineers with little software dev experience. But we want to test the network with mechanical energy harvesters on the nRF 52 dev kit.

We came across a strange thing today. It seems we only can set the connection interval to a max value of 400 ms as it seem to crash the soft-device (S132 v2.0.0) if we set it to a higher value. We are using the build from issue #77 to get a working mesh on the Nrf52 DK. So we have a hard time researching the best tradeoff between performance and energy balance for our application.

Could this be connected to the thing you mention about "The slave connection procedure still has problems with high intervals as of now." in the battery test section in the wiki?

Our slave_latency = 4 
Max_connection_interval = 1000
conn_sup_timeout = 6000

All other setting is the same as in the config.h from the working mesh application from issue #77, but as described earlier it seems to crash with the soft device somehow.

I should work within the connection constraints described in BT Spec 4.1 Vol 2 Part E, Section 7.8.12 
Or as what you describe in the config file as a constraint:
conn_sup_timeout * 4 > (1 + slave_latency) * max_conn_interval

Dving16, 2017-11-15:
Ok.  It seems like setting the slave latency to 0 did the job. Now the soft device don´t crash. But when we set connection_interval to more than 400 ms, handshake is timeout when nodes are trying to make connection.

mariusheil, 2019-03-05:
closing this as it was rather old. Should be reevaluated with the latest release

closed 2019-03-05 by mariusheil