bluerange-io/bluerange-mesh #98: how to use AppConnections ?

alanzheng800, 2018-02-27:
Your mesh stack is very good! I want to use it in my product. but,how to use a phone connect to it .If you have a example,please contact me !

mariusheil, 2018-03-19:
Hi,

sorry for the late comment. I am working to get a new version to Github, but there is not enough free time right now to clean it up and publish it. What are you planning to do?
You can inherit from the AppConnection and use the ResolverConnection to detect the connection type. Maybe if you have a look, you will get the idea and I can help you with some more specific question as I currently do not have an example that I could give you.

Marius

OSPro, 2018-04-09:
Hi, i have same question. For example, how i send command to all nodes inside network from Mobile APP like Android?

mariusheil, 2019-03-05:
Take a  look at the new MeshAccessConnections which is a general aproach on how to communicate with an app. We have an iOS implementation for that but there is now timeline on when this goes open source yet.

closed 2019-03-05 by mariusheil