bluerange-io/bluerange-mesh #102: Is fruitymesh still being developed?

romankc, 2019-01-06:
I see that about a year ago you were looking to hire somebody to work on fruitymesh and that there have been no updates for over a year. Is this project still actively being developed?

volkrass, 2019-02-18:
Dear @romankc, yes it is still under heavy development. 
A brand new version with a lot of improvements will be available in few days. Stay tuned!

romankc, 2019-02-18:
We recently started building a product with the regular Nordic Mesh SDK so I am excited to try your implementation instead. Certainly willing to pay reasonable licensing fees for advanced features.

volkrass, 2019-02-20:
@romankc you can reach me out at https://bluerange.io/contact/

volkrass, 2019-03-05:
New version is out.

closed 2019-03-05 by volkrass