bluerange-io/bluerange-mesh #103: PuTtY does not have any output

mabner1996, 2019-03-22:
Hello, I tried following the Quick Start guide.
I am using nrf51 development board,
I already compile the Fruitymesh and got the Fruitymesh.hex file

Then, I flash the device through nrf-connect programmer app by adding both softdevice (I use 110 softdevice) and fruitymesh .hex files into the chip.

I tried to use putty with the configuration specified in the quick start guide.
However after the terminal connected and I pushed the reset button, PuTtY does not show any output.

where do I do wrong?

mariusheil, 2019-03-22:
Hi, are you using the nrf51-DK and not the dongle? Do you see the leds flashing after you flashed the hex files? If no LEDs are flashing, then probably something went wrong during programming. The provided release contains the softdevice already. Try to erase the chip again, then only program the provided binary release. Does that work? If yes, you can try your own compiled version. Can you try to use the makefile for flashing? It is explained in the guide as well how to flash using the makefile.

mabner1996, 2019-03-25:
I tried erasing the chip and flash just the binary using nrf connect. It managed to flash and nrf connect can read the device.
However, PuTtY still doesn't show any response after I hit the reset button in my device.

I tried erasing the chip again and using the command prompt ("make" and then "make flash").
It has no error, however, no output is shown again in the PuTtY.

I use a profilic serial cable in my com port 5 for this.

mariusheil, 2019-03-25:
Hello,

if you use the nrf51-DK, you should not need to use a different serial cable. The pins are configured to be routed to the onboard segger chip which gives you a virtual COM port to read from. Just connect the nrf51-dk via USB and check if you have a virtual com port in your device manager. This one should work.

mabner1996, 2019-03-26:
hello, this is the picture of my nrf 51 development kit
the black cable is the serial profilic cable which I mentioned in my last comment, it connected to port 5 in my com. But as I told before, putty doesnt show any output when I press restart.

In addition, no LED are flashing when I flash the program

for debug, usually I connect it with j-link edu from SEGGER

[image]

mariusheil, 2019-03-26:
Hello,

that's not an nrf51-DK ;-) That's the one that is mentioned in the documentation: https://www.nordicsemi.com/Software-and-Tools/Development-Kits/nRF51-DK For your board, you need to specify different pins, it is probably not pin compatible. Check the documentation of your board for the led and the uart RX/TX/RTS/CTS pins. Go to /config/boards and open board_0.cpp. For now the easiest way is to modify that file. Check all pin settings and make sure the uart pins are correct (sometimes rx/tx has to be interchanged. Same with rts/cts). You can also use the Segger RTT Viewer. Open it up, connect via USB, select the correct chipset, then configure:
- input => sending => send on enter
- input => end of line => mac format
This should also give you a terminal over the segger debugger.

mabner1996, 2019-03-27:
Hello, Thank you for your reply
I already get the documentation of the board I am using. (I attached the pdf of the config)
[ble board.pdf](https://github.com/mwaylabs/fruitymesh/files/3011973/ble.board.pdf)
So I make several changes to config to adjust it with the documentation
1. LED PIN 1 from 21 to 16,
2. LED PIN 2 from 22 to 17, 
3. LED PIN 3 from 23 to 18
4. swap the RTS CTS pin of UART (CTS = 08 RTS = 10)

As I saved it, I tried to use commanrd prompt to compile and flash the program
The led light in the board still not flash, and the terminal (tried use both PUTTY and JLINK RTT Viewer) still doesnt have any output

mabner1996, 2019-04-03:
Actually, I used 16 kB RAM nrf 51822 chip which is wrong. So I tried to flash it onto 32 kB RAM nrf 51822 chip.
Now RTT Viewer gives some output which I attached the screenshot here. 
[image]

However, it does not show anything if I use PuTtY.
Do you know any possible cause of this problem?

Thank you

mariusheil, 2019-04-03:
Looks like it is now booting. Try to enter some commands in the segger terminal and see if they work. Try to switch the led on and off so you can see if your pin config is correct. If you do not get output in putty, you have a wrong serial port configuration, so either your cable is problematic, the pins are incorrect, baud rate or flow control is wrong. You could also have the serial settings wrong in putty. Once you have the led working, you can only read through the specification again and try to play around a little with the settings

mabner1996, 2019-04-04:
Hello, actually I forgot to mention that I only connect 4 pin only from the nrf51chip to the development kit which are VCC, GND, SWDIO, and SWCLK
with this 4 pin configuration, we can properly flashed the fruitymesh and view some output (that I screenshot earlier) using RTT Viewer.
Is it possible that for the program to be seen in serial port (in this case, PuTtY) we need to connect more pins than these 4??

mariusheil, 2019-04-10:
For UART, you need extra pins. You need to have some kind of USB => Serial adapter attached. Uart needs at least VCC, GND, TX and RX lines and additionally RTS/CTS lines if you are using flow control.

mabner1996, 2019-04-11:
Okay, thanks a lot. We already plugged it the 4 extra pins (RXD TXD RTS CTS) and place it according to our board and chip documentation.
We find out that we also need to also do this following two things to get the message:
1. In Terminal.cpp file, line 68. We changed the setting of terminalIsInitialized = false into true
2. In serial port terminal settings (PuTtY) we set the data bits into 7 instead of 8, the rest of the configurations are still following the quick start guide.

Right now, we managed to see some output in PuTtY. However, when we input any command, including "status", "reset", and "data" on the terminal, it always say "command not found".
However, those commands can be entered and processed through RTTViewer normally.

Right now we tried to debugging this problem. Once we have new information, we will post it here.

mabner1996, 2019-04-15:
Thanks a lot for your help.
The terminal now has output and now we can put our command after switching to another development kit that supports 32 kB RAM NRF51822 chip. However, one thing should be noted that our development kit does not support 10000000 baud rate. Thus we need to lower the speed (right now we are using 38400 baud rate).

closed 2019-04-15 by mabner1996