bluerange-io/bluerange-mesh #104: Nrf52840 support

giowild, 2019-04-11:
Is nrf52840 supported?
Is Softdevice S140 supported?
Is Bluetooth 5 long range phy coded supported?

Thanks and keep up with your good work

mariusheil, 2019-04-11:
yes, yes, no ;-)
Currently, the SoftDevice 5 alpha should work. Has not yet been ported to softdevice 6. Probably not a lot of work. With v6, a lot of new features were added, so if you need production ready nrf52840 support. you should probably try porting, or we will do that in the weeks to come. I guess that not a lot of work is needed to support long range. But some tweaking of the parameters and some thinking about the use-case should probably be done.

giowild, 2019-04-12:
Ok, thank you a lot :)

One last question: is it suited for sensors monitoring too or is it better for configuring set o beacons?

mariusheil, 2019-04-12:
Sensor monitoring exactly what we are using it for ;-) It's of course also possible to use it to configure beacons or attached actors / 3rd party devices. Most of our use-cases involve attached sensors / actors or 3rd party devices.

closed 2019-04-12 by mariusheil

giowild, 2019-04-12:
Hi,
I tried compiling with Eclipse+GCC and it works well when using NRF51 or NRF52 in Makefile.local.

But when if I use NRF52840, it crashes with this error

```
make VERBOSE=1 -j4 all 
make: *** No rule to make target '_build/release/NRF52840/github/spi.o', needed by '_build/release/NRF52840/github/FruityMesh.out'.  Stop.
"make VERBOSE=1 -j4 all" terminated with exit code 2. Build might be incomplete.
````

Commenting the following lines in Makefile.NRF52840 did the job, but what if I need spi or bme280? Where can I find the needed files, given that these are not included in the repository? 
```
C_SOURCE_FILES += src/c/drivers/spi.c
C_SOURCE_FILES += src/c/drivers/bme280.c
C_SOURCE_FILES += src/c/drivers/lis2dh12.c
C_SOURCE_FILES += src/c/drivers/nrf_drv_spi.c
```

marcantoinebouchard, 2020-04-15:
Hi,

Have you been able to make it work with Bluetooth 5 long range?

Thanks